LibreDWG/libredwg #693: Wrong encoding of tables from json to DWG in preR13

michal-josef-spacek, 2023-04-11:
Example:
[test.json.gz](https://github.com/LibreDWG/libredwg/files/11202163/test.json.gz)
Command: ```dwgwrite test.json```
Output:
```
...
Skip object BLOCK_HEADER, number: 5, Fixedtype: 49, Addr: 0 (0x2e5)
Skip object LAYER, number: 6, Fixedtype: 51, Addr: 0 (0x2e5)
Skip object STYLE, number: 7, Fixedtype: 53, Addr: 0 (0x2e5)
Skip object LTYPE, number: 8, Fixedtype: 57, Addr: 0 (0x2e5)

entities 0 0x2e5 - 0x2e5

ctrl BLOCK_CONTROL [0]: num:0

ctrl LAYER_CONTROL [0]: num:1
contents table LAYER [0]: size:37 (0x0, 0x2e5)
ERROR: Wrong type BLOCK_CONTROL, expected DWG_TYPE_LAYER

ctrl STYLE_CONTROL [0]: num:1
contents table STYLE [0]: size:37 (0x0, 0x2e5)
ERROR: Wrong type BLOCK_CONTROL, expected DWG_TYPE_STYLE

ctrl LTYPE_CONTROL [0]: num:1
contents table LTYPE [0]: size:37 (0x0, 0x2e5)
ERROR: Wrong type BLOCK_CONTROL, expected DWG_TYPE_LTYPE

ctrl VIEW_CONTROL [0]: num:0
block_entities   0x2e5 - 0x2e5 (0x40000000)
@0x14:
entities_start: 0x2e5 [RLx]
...
```

rurban, 2023-04-12:
Not just encode, also json. address and num is also missing.

rurban, 2023-04-12:
This is caused by wrong handle ref resolving. We need to add the obj->index as absolute_ref to r11 handles

rurban, 2023-04-14:
I believe this is fixed now.
Just json's are incompatible now

michal-josef-spacek, 2023-04-14:
Maybe yes

closed 2023-04-15 by rurban