LibreDWG/libredwg #694: Segmentation fault in dwg_decode__3DSOLID_private

fmunozs, 2023-04-16:
Attached file cases a segmentation fault on Debian 11 x86 with the current GIT release.

[1.zip](https://github.com/LibreDWG/libredwg/files/11244854/1.zip)

```
user@debian:~/libredwg$ gdb ./dwgread
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
(gdb)
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "i686-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./dwgread...
(gdb) r -v9 /home/user/out/default/crashes/1*
Starting program: /home/user/libredwg/dwgread -v9 /home/user/out/default/crashes/1*
Reading DWG file /home/user/out/default/crashes/1.dwg

Program received signal SIGSEGV, Segmentation fault.
0xb6cfb56d in dwg_decode__3DSOLID_private (dat=<optimized out>, hdl_dat=<optimized out>, str_dat=0xbfffdd48, obj=0x825bab0) at src/dwg.spec:3001
3001      ACTION_3DSOLID;
(gdb)
```

rurban, 2023-04-18:
Oh sorry. Didnt see the x86 hint

rurban, 2023-04-18:
Indeed a new type of error. c/malloc sizes may not exceed maximum supported size of 0xc0000000 on 32bit. Limit this for 64bit also, where would have max 0xffffffff for BL sized vectors.

closed 2023-04-18 by rurban