LibreDWG/libredwg #695: Segmentation fault in acds_private

fmunozs, 2023-04-16:
[2.zip](https://github.com/LibreDWG/libredwg/files/11244857/2.zip)

```
(gdb)
(gdb) r -v9 /home/user/out/default/crashes/2*
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/user/libredwg/dwgread -v9 /home/user/out/default/crashes/2*
Reading DWG file /home/user/out/default/crashes/2.dwg

Program received signal SIGSEGV, Segmentation fault.
0xb6c8eb26 in acds_private (dat=0xbfffe330, dwg=0xbfffe578) at src/acds.spec:215
215                 SUB_FIELD_RL (schidx.props[rcount2],index, 0);
(gdb)

```

rurban, 2023-04-17:
Not repro with c8e648437aab4d6a26af35531a45e07a2bfacf21

fmunozs, 2023-04-17:
Still segfaults for me on 32 bits VM debian 11 machine.

```
user@debian:~/repro/libredwg$ valgrind ./dwgread /home/user/out/default/crashes/2.dwg
==19822== Memcheck, a memory error detector
==19822== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==19822== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==19822== Command: ./dwgread /home/user/out/default/crashes/2.dwg
==19822==
Warning: checksum: 0x28dd125e (calculated) mismatch

Warning: Invalid sections: 0 != numgaps: 0 + numsections: 16
ERROR: Invalid lit_length 16 > 10 bytes left
Warning: Invalid handle-reference, longer than 8 bytes: 0.11.0
ERROR: bit_read_BD: unexpected 2-bit code: '11'
ERROR: Invalid BD _3DDWFPREC
Warning: Unknown Class object 500 ጁACDBDICTIONARYWDFLT卄ᣀ᪀ᦀᡀ᷀ᄀက ࠀကᯀ᠀᱀᳀ᧀ᱀Ⴠჴ〆䀄 쀆ဆ〆倆耄쀆䀆倆 㴄ĄČĐĈŀİĄČĔĠļİĐĔň休戀樀攀挀琀䐀䈀堀 䌀氀愀猀猀攀猀䈀䂐 (0x0)
Warning: check_CRC mismatch 11945-11976 = 31: 6853 <=> 9514

...
ERROR: bit_read_RC buffer overflow at 30.3 + 1 > 31
ERROR: bit_read_BB buffer overflow at 31.1 > 31
ERROR: bit_read_BL buffer overflow at 31.1 > 31
ERROR: bit_read_BB buffer overflow at 31.1 > 31
ERROR: bit_read_BS buffer overflow at 31.1 > 31
ERROR: bit_read_CMC buffer overflow at 31.1 + 1 > 31
ERROR: bit_read_BB buffer overflow at 31.1 > 31
ERROR: bit_read_BD buffer overflow at 31.1 > 31
ERROR: Invalid BD bend_line_length
Warning: check_CRC mismatch 18610-18660 = 50: 5E5B <=> 8376

Warning: check_CRC mismatch 18662-18712 = 50: D5E5 <=> 08C8

Warning: check_CRC mismatch 18558-18608 = 50: 559C <=> 88B1

ERROR: Invalid zero_18 size 3. Need min. 48 bits, have 65280 for .
Warning: thumbnail.size mismatch: 4366 != 0
==19822== Invalid write of size 4
==19822==    at 0x4E8C506: decode_R2004 (in /home/user/repro/libredwg/libredwg.so)
==19822==    by 0x4E95166: dwg_decode (in /home/user/repro/libredwg/libredwg.so)
==19822==    by 0x4EF4ECF: dwg_read_file (in /home/user/repro/libredwg/libredwg.so)
==19822==    by 0x10978A: main (in /home/user/repro/libredwg/dwgread)
==19822==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==19822==
==19822==
==19822== Process terminating with default action of signal 11 (SIGSEGV)
==19822==  Access not within mapped region at address 0x0
==19822==    at 0x4E8C506: decode_R2004 (in /home/user/repro/libredwg/libredwg.so)
==19822==    by 0x4E95166: dwg_decode (in /home/user/repro/libredwg/libredwg.so)
==19822==    by 0x4EF4ECF: dwg_read_file (in /home/user/repro/libredwg/libredwg.so)
==19822==    by 0x10978A: main (in /home/user/repro/libredwg/dwgread)
==19822==  If you believe this happened as a result of a stack
==19822==  overflow in your program's main thread (unlikely but
==19822==  possible), you can try to increase the size of the
==19822==  main thread stack using the --main-stacksize= flag.
==19822==  The main thread stack size used in this run was 8388608.
==19822==
==19822== HEAP SUMMARY:
==19822==     in use at exit: 269,467 bytes in 1,770 blocks
==19822==   total heap usage: 1,913 allocs, 143 frees, 577,454 bytes allocated
==19822==
==19822== LEAK SUMMARY:
==19822==    definitely lost: 0 bytes in 0 blocks
==19822==    indirectly lost: 0 bytes in 0 blocks
==19822==      possibly lost: 0 bytes in 0 blocks
==19822==    still reachable: 269,467 bytes in 1,770 blocks
==19822==         suppressed: 0 bytes in 0 blocks
==19822== Rerun with --leak-check=full to see details of leaked memory
==19822==
==19822== For lists of detected and suppressed errors, rerun with: -s
==19822== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault

```

closed 2023-04-18 by rurban