LibreDWG/libredwg #702: Wrong conversion of block entities from JSON to DWG in preR13

michal-josef-spacek, 2023-04-22:
Input JSON file:
[test.json.gz](https://github.com/LibreDWG/libredwg/files/11300931/test.json.gz)

command: ```dwgwrite test.json```
Output:
```
...
Entities from 10 to 12 from 0x465
===========================
Entity BLOCK, number: 10, Addr: 1125 (0x465)
type: 12 [RC]
Encode entity BLOCK
flag_r11: 0x0 [RC 0] @2.0
size: 24 [RS] @4.0
layer: 0 [RSd 8]
opts_r11: 0x0 [RSx 0] @8.0
base_pt: (0.000000, 0.000000) [2RD 10] @24.0
===========================
Entity LINE, number: 11, Addr: 1149 (0x47d)
type: 1 [RC]
Encode entity LINE
flag_r11: 0x0 [RC 0] @2.0
size: 40 [RS] @4.0
layer: 0 [RSd 8]
opts_r11: 0x0 [RSx 0] @8.0
start: (0.000000, 0.000000) [2RD 10] @24.0
end: (1.000000, 1.000000) [2RD 11] @40.0
block_entities   0x465 - 0x4a5 (0x40)
@0x14:
entities_start: 0x2e5 [RLx]
entities_end: 0x2e5 [RLx]
blocks_start: 0x465 [RLx]
blocks_size: 0x40 [RLx]
extras_start: 0x0 [RLx]
extras_size: 0x80000000 [RLx]
numentities: 0 [RS]
Wrote 0 bytes
...
```
**There is missing last ENDBLK entity.**

Output file has block entities like:
```
...
blocks entities: (0x465-0x4a5 (0), size 64)
==========================================
Warning:
@0x46c => start 0x465
==========================================
blocks entities: end
...
```

michal-josef-spacek, 2023-04-22:
The same issue is in AC1.40.
Example:
DWG file: [test2.dwg.gz](https://github.com/LibreDWG/libredwg/files/11301015/test2.dwg.gz)
JSON from test2 DWG file: [test2.json.gz](https://github.com/LibreDWG/libredwg/files/11301017/test2.json.gz)

The issue is with the conversion from JSON to DWG.

michal-josef-spacek, 2023-04-22:
Same issue is with dwgadd and block.

michal-josef-spacek, 2023-04-24:
@rurban I will test it.

btw:
```
encode.c: In function ‘dwg_encode’:
encode.c:2548:10: warning: ‘first_block’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 2548 |       if (first_block)
      |          ^
```

michal-josef-spacek, 2023-04-24:
dwgadd is working

The original issue from JSON to DWG is not working. There are informations about entities in log file, but not in final DWG file.
Probably bad ```blocks_start: 0x465 [RLx 0]``` Should be ```0x46c```

michal-josef-spacek, 2023-04-25:
There is issue with numentities header variable. It contain number of entities in entities section, but no in block and extra section. Variable contain sum of all entities.

rurban, 2023-05-08:
The block entities are now all in.
For the HEADER.numentities I'll check

closed 2023-05-08 by rurban