LibreDWG/libredwg #703: Conditional jump or move depends on uninitialised value dwg_free_BLOCK_CONTROL_private

fmunozs, 2023-04-23:
Tested this on a 32 bits VM.

[dwg_free_BLOCK_CONTROL_private.zip](https://github.com/LibreDWG/libredwg/files/11304223/dwg_free_BLOCK_CONTROL_private.zip)

```
user@debian:~/libredwg$ valgrind ./dwgread dwg_free_BLOCK_CONTROL_private.dwg
==9269== Memcheck, a memory error detector
==9269== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==9269== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==9269== Command: ./dwgread dwg_free_BLOCK_CONTROL_private.dwg
==9269==
==9269== Conditional jump or move depends on uninitialised value(s)
==9269==    at 0x44301BA: dwg_free_BLOCK_CONTROL_private (dwg.spec:3473)
==9269==    by 0x43E35D4: dwg_free_BLOCK_CONTROL (dwg.spec:3461)
==9269==    by 0x444AF3F: free_preR13_object (free.c:803)
==9269==    by 0x444AF3F: dwg_free_object (free.c:911)
==9269==    by 0x4451A20: dwg_free (free.c:1619)
==9269==    by 0x804B140: main (dwgread.c:336)
==9269==
ERROR 0x801
==9269==
==9269== HEAP SUMMARY:
==9269==     in use at exit: 0 bytes in 0 blocks
==9269==   total heap usage: 47 allocs, 47 frees, 110,885 bytes allocated
==9269==
==9269== All heap blocks were freed -- no leaks are possible
==9269==
==9269== Use --track-origins=yes to see where uninitialised values come from
==9269== For lists of detected and suppressed errors, rerun with: -s
==9269== ERROR SUMMARY: 37 errors from 1 contexts (suppressed: 0 from 0)
```

rurban, 2023-04-24:
Duplicate of #701

rurban, 2023-04-24:
Fixed with b0af2cc31fb7a1b4099a804e1729344befc77f14

fmunozs, 2023-04-25:
Hello @rurban 

I have just tested again with the current master and the same warning happens on Valgrind:

```
user@debian:~/clang/libredwg$ valgrind ./dwgread ./dwg_free_BLOCK_CONTROL_private.dwg
==6801== Memcheck, a memory error detector
==6801== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==6801== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==6801== Command: ./dwgread ./dwg_free_BLOCK_CONTROL_private.dwg
==6801==
==6801== Conditional jump or move depends on uninitialised value(s)
==6801==    at 0x4433F0A: dwg_free_BLOCK_CONTROL_private (dwg.spec:3473)
==6801==    by 0x43E7324: dwg_free_BLOCK_CONTROL (dwg.spec:3461)
==6801==    by 0x444EC8F: free_preR13_object (free.c:803)
==6801==    by 0x444EC8F: dwg_free_object (free.c:911)
==6801==    by 0x4455770: dwg_free (free.c:1619)
==6801==    by 0x804B140: main (dwgread.c:336)
==6801==
ERROR 0x801
==6801==
==6801== HEAP SUMMARY:
==6801==     in use at exit: 0 bytes in 0 blocks
==6801==   total heap usage: 47 allocs, 47 frees, 110,885 bytes allocated
==6801==
==6801== All heap blocks were freed -- no leaks are possible
==6801==
==6801== Use --track-origins=yes to see where uninitialised values come from
==6801== For lists of detected and suppressed errors, rerun with: -s
==6801== ERROR SUMMARY: 37 errors from 1 contexts (suppressed: 0 from 0)
```