LibreDWG/libredwg #704: Invalid write of size 4 dwg_decode_POLYLINE_3D_private

fmunozs, 2023-04-23:
Tested on a 32 bit machine. 

[dwg_decode_POLYLINE_3D_private.zip](https://github.com/LibreDWG/libredwg/files/11304235/dwg_decode_POLYLINE_3D_private.zip)

```
user@debian:~/libredwg$ valgrind ./dwgread dwg_decode_POLYLINE_3D_private.dwg
==10391== Memcheck, a memory error detector
==10391== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==10391== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==10391== Command: ./dwgread dwg_decode_POLYLINE_3D_private.dwg
==10391==
==10391== Invalid write of size 4
==10391==    at 0x42DE333: dwg_decode_POLYLINE_3D_private (dwg.spec:0)
==10391==    by 0x42C0DE7: dwg_decode_POLYLINE_3D (dwg.spec:0)
==10391==    by 0x42BDCD1: dwg_decode_add_object (decode.c:5257)
==10391==    by 0x428015D: read_2004_section_handles (decode.c:2541)
==10391==    by 0x428015D: decode_R2004 (decode.c:3426)
==10391==    by 0x427BE40: dwg_decode (decode.c:0)
==10391==    by 0x439018F: dwg_read_file (dwg.c:261)
==10391==    by 0x804ACDA: main (dwgread.c:0)
==10391==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==10391==
==10391==
==10391== Process terminating with default action of signal 11 (SIGSEGV)
==10391==  Access not within mapped region at address 0x0
==10391==    at 0x42DE333: dwg_decode_POLYLINE_3D_private (dwg.spec:0)
==10391==    by 0x42C0DE7: dwg_decode_POLYLINE_3D (dwg.spec:0)
==10391==    by 0x42BDCD1: dwg_decode_add_object (decode.c:5257)
==10391==    by 0x428015D: read_2004_section_handles (decode.c:2541)
==10391==    by 0x428015D: decode_R2004 (decode.c:3426)
==10391==    by 0x427BE40: dwg_decode (decode.c:0)
==10391==    by 0x439018F: dwg_read_file (dwg.c:261)
==10391==    by 0x804ACDA: main (dwgread.c:0)
==10391==  If you believe this happened as a result of a stack
==10391==  overflow in your program's main thread (unlikely but
==10391==  possible), you can try to increase the size of the
==10391==  main thread stack using the --main-stacksize= flag.
==10391==  The main thread stack size used in this run was 8388608.
==10391==
==10391== HEAP SUMMARY:
==10391==     in use at exit: 614,149 bytes in 2,757 blocks
==10391==   total heap usage: 2,832 allocs, 75 frees, 732,560 bytes allocated
==10391==
==10391== LEAK SUMMARY:
==10391==    definitely lost: 0 bytes in 0 blocks
==10391==    indirectly lost: 0 bytes in 0 blocks
==10391==      possibly lost: 0 bytes in 0 blocks
==10391==    still reachable: 614,149 bytes in 2,757 blocks
==10391==         suppressed: 0 bytes in 0 blocks
==10391== Rerun with --leak-check=full to see details of leaked memory
==10391==
==10391== For lists of detected and suppressed errors, rerun with: -s
==10391== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault
```

rurban, 2023-04-24:
Duplicate of #701

rurban, 2023-04-24:
Fixed with b0af2cc31fb7a1b4099a804e1729344befc77f14, errors now earlier

closed 2023-04-24 by rurban