LibreDWG/libredwg #705: Invalid write of size 4 dwg_decode_INSERT_private

fmunozs, 2023-04-23:
[dwg_decode_INSERT_private.zip](https://github.com/LibreDWG/libredwg/files/11304246/dwg_decode_INSERT_private.zip)

```
user@debian:~/libredwg$ valgrind ./dwgread dwg_decode_INSERT_private.dwg
==10412== Memcheck, a memory error detector
==10412== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==10412== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==10412== Command: ./dwgread dwg_decode_INSERT_private.dwg
==10412==
==10412== Invalid write of size 4
==10412==    at 0x42DAD84: dwg_decode_INSERT_private (dwg.spec:0)
==10412==    by 0x42C0027: dwg_decode_INSERT (dwg.spec:0)
==10412==    by 0x42BDB75: dwg_decode_add_object (decode.c:5231)
==10412==    by 0x428015D: read_2004_section_handles (decode.c:2541)
==10412==    by 0x428015D: decode_R2004 (decode.c:3426)
==10412==    by 0x427BE40: dwg_decode (decode.c:0)
==10412==    by 0x439018F: dwg_read_file (dwg.c:261)
==10412==    by 0x804ACDA: main (dwgread.c:0)
==10412==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==10412==
==10412==
==10412== Process terminating with default action of signal 11 (SIGSEGV)
==10412==  Access not within mapped region at address 0x0
==10412==    at 0x42DAD84: dwg_decode_INSERT_private (dwg.spec:0)
==10412==    by 0x42C0027: dwg_decode_INSERT (dwg.spec:0)
==10412==    by 0x42BDB75: dwg_decode_add_object (decode.c:5231)
==10412==    by 0x428015D: read_2004_section_handles (decode.c:2541)
==10412==    by 0x428015D: decode_R2004 (decode.c:3426)
==10412==    by 0x427BE40: dwg_decode (decode.c:0)
==10412==    by 0x439018F: dwg_read_file (dwg.c:261)
==10412==    by 0x804ACDA: main (dwgread.c:0)
==10412==  If you believe this happened as a result of a stack
==10412==  overflow in your program's main thread (unlikely but
==10412==  possible), you can try to increase the size of the
==10412==  main thread stack using the --main-stacksize= flag.
==10412==  The main thread stack size used in this run was 8388608.
==10412==
==10412== HEAP SUMMARY:
==10412==     in use at exit: 8,736,970 bytes in 76,745 blocks
==10412==   total heap usage: 84,987 allocs, 8,242 frees, 24,663,830 bytes allocated
==10412==
==10412== LEAK SUMMARY:
==10412==    definitely lost: 0 bytes in 0 blocks
==10412==    indirectly lost: 0 bytes in 0 blocks
==10412==      possibly lost: 0 bytes in 0 blocks
==10412==    still reachable: 8,736,970 bytes in 76,745 blocks
==10412==         suppressed: 0 bytes in 0 blocks
==10412== Rerun with --leak-check=full to see details of leaked memory
==10412==
==10412== For lists of detected and suppressed errors, rerun with: -s
==10412== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault
```

rurban, 2023-04-24:
Duplicate of #701

rurban, 2023-04-24:
Fixed with b0af2cc31fb7a1b4099a804e1729344befc77f14