LibreDWG/libredwg #706: Invalid write of size 4 dwg_decode_MINSERT_private

fmunozs, 2023-04-23:
[dwg_decode_MINSERT_private.zip](https://github.com/LibreDWG/libredwg/files/11304268/dwg_decode_MINSERT_private.zip)

I think this could be related to #705 as I see some similarity on the call stack.
```
user@debian:~/libredwg$ valgrind ./dwgread dwg_decode_MINSERT_private.dwg
==10452== Memcheck, a memory error detector
==10452== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==10452== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==10452== Command: ./dwgread dwg_decode_MINSERT_private.dwg
==10452==
==10452== Invalid write of size 4
==10452==    at 0x42DBDF4: dwg_decode_MINSERT_private (dwg.spec:0)
==10452==    by 0x42C0187: dwg_decode_MINSERT (dwg.spec:0)
==10452==    by 0x42BDB9F: dwg_decode_add_object (decode.c:5234)
==10452==    by 0x428015D: read_2004_section_handles (decode.c:2541)
==10452==    by 0x428015D: decode_R2004 (decode.c:3426)
==10452==    by 0x427BE40: dwg_decode (decode.c:0)
==10452==    by 0x439018F: dwg_read_file (dwg.c:261)
==10452==    by 0x804ACDA: main (dwgread.c:0)
==10452==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==10452==
==10452==
==10452== Process terminating with default action of signal 11 (SIGSEGV)
==10452==  Access not within mapped region at address 0x0
==10452==    at 0x42DBDF4: dwg_decode_MINSERT_private (dwg.spec:0)
==10452==    by 0x42C0187: dwg_decode_MINSERT (dwg.spec:0)
==10452==    by 0x42BDB9F: dwg_decode_add_object (decode.c:5234)
==10452==    by 0x428015D: read_2004_section_handles (decode.c:2541)
==10452==    by 0x428015D: decode_R2004 (decode.c:3426)
==10452==    by 0x427BE40: dwg_decode (decode.c:0)
==10452==    by 0x439018F: dwg_read_file (dwg.c:261)
==10452==    by 0x804ACDA: main (dwgread.c:0)
==10452==  If you believe this happened as a result of a stack
==10452==  overflow in your program's main thread (unlikely but
==10452==  possible), you can try to increase the size of the
==10452==  main thread stack using the --main-stacksize= flag.
==10452==  The main thread stack size used in this run was 8388608.
==10452==
==10452== HEAP SUMMARY:
==10452==     in use at exit: 546,478 bytes in 704 blocks
==10452==   total heap usage: 1,115 allocs, 411 frees, 658,870 bytes allocated
==10452==
==10452== LEAK SUMMARY:
==10452==    definitely lost: 0 bytes in 0 blocks
==10452==    indirectly lost: 0 bytes in 0 blocks
==10452==      possibly lost: 0 bytes in 0 blocks
==10452==    still reachable: 546,478 bytes in 704 blocks
==10452==         suppressed: 0 bytes in 0 blocks
==10452== Rerun with --leak-check=full to see details of leaked memory
==10452==
==10452== For lists of detected and suppressed errors, rerun with: -s
==10452== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault
```

rurban, 2023-04-24:
Duplicate of #701

rurban, 2023-04-24:
Fixed with b0af2cc31fb7a1b4099a804e1729344befc77f14

closed 2023-04-24 by rurban