LibreDWG/libredwg #708: Invalid read of size 4 memcpy bit_read_fixed

fmunozs, 2023-04-23:
[bit_read_fixed.zip](https://github.com/LibreDWG/libredwg/files/11304306/bit_read_fixed.zip)

```
user@debian:~/libredwg$ valgrind ./dwgread bit_read_fixed.dwg
==10557== Memcheck, a memory error detector
==10557== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==10557== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==10557== Command: ./dwgread bit_read_fixed.dwg
==10557==
==10557== Invalid read of size 4
==10557==    at 0x403B56D: memcpy (vg_replace_strmem.c:1034)
==10557==    by 0x426F92B: bit_read_fixed (bits.c:1585)
==10557==    by 0x426F9FA: bit_read_TF (bits.c:1614)
==10557==    by 0x42CDE7E: decode_preR13_sentinel (decode.c:6223)
==10557==    by 0x4384061: decode_preR13_section (decode_r11.c:0)
==10557==    by 0x43809C8: decode_preR13 (decode_r11.c:752)
==10557==    by 0x427BB22: dwg_decode (decode.c:217)
==10557==    by 0x439018F: dwg_read_file (dwg.c:261)
==10557==    by 0x804ACDA: main (dwgread.c:0)
==10557==  Address 0x54d92a8 is 16 bytes before a block of size 4,849 alloc'd
==10557==    at 0x4038A16: calloc (vg_replace_malloc.c:760)
==10557==    by 0x438FAEB: dat_read_file (dwg.c:95)
==10557==    by 0x4390093: dwg_read_file (dwg.c:254)
==10557==    by 0x804ACDA: main (dwgread.c:0)
==10557==
==10557== Invalid read of size 4
==10557==    at 0x403B57B: memcpy (vg_replace_strmem.c:1034)
==10557==    by 0x426F92B: bit_read_fixed (bits.c:1585)
==10557==    by 0x426F9FA: bit_read_TF (bits.c:1614)
==10557==    by 0x42CDE7E: decode_preR13_sentinel (decode.c:6223)
==10557==    by 0x4384061: decode_preR13_section (decode_r11.c:0)
==10557==    by 0x43809C8: decode_preR13 (decode_r11.c:752)
==10557==    by 0x427BB22: dwg_decode (decode.c:217)
==10557==    by 0x439018F: dwg_read_file (dwg.c:261)
==10557==    by 0x804ACDA: main (dwgread.c:0)
==10557==  Address 0x54d92ac is 12 bytes before a block of size 4,849 alloc'd
==10557==    at 0x4038A16: calloc (vg_replace_malloc.c:760)
==10557==    by 0x438FAEB: dat_read_file (dwg.c:95)
==10557==    by 0x4390093: dwg_read_file (dwg.c:254)
==10557==    by 0x804ACDA: main (dwgread.c:0)
...
```

rurban, 2023-04-24:
Duplicate of #701

rurban, 2023-04-24:
Fixed by stricter wrong section reading. Fails now earlier on !number, !size or invalid sentinel

closed 2023-04-24 by rurban