LibreDWG/libredwg #709: Wrong parsing of R11 tables without items in them

michal-josef-spacek, 2023-04-24:
This commit https://github.com/LibreDWG/libredwg/commit/b0af2cc31fb7a1b4099a804e1729344befc77f14 is wrong.
We need to parse sentinels each time.

Example:
[BLANK.DWG.gz](https://github.com/LibreDWG/libredwg/files/11312586/BLANK.DWG.gz)

Bad output:
```
...
entities: (0x6cf-0x6cf (0), size 0)
==========================================
DWG_SENTINEL_R11_ENTITIES_BEGIN: C46E6854F86E3330633EC1852ADC9401
DWG_SENTINEL_R11_ENTITIES_END: 3B9197AB0791CCCF9CC13E7AD5236BFE
==========================================
entities: end

contents table BLOCK    [ 1]: size:45   num:0   (0x6ef-0x6ef)

contents table LAYER    [ 2]: size:41   num:1   (0x70f-0x738)

Warning:
@0x6df => start 0x6ff
unknown (32): DBEFB3F0C73E6DA6C9B6245C4C6F32CB24104C0F38C192593649DBA3B390CD34
DWG_SENTINEL_R11_LAYERS_BEGIN: 0EC4646FBB1DD38B0049C2EF18EA6FFB
...
```

In unknown (32) are begin and end sentinels of BLOCK table.

rurban, 2023-04-25:
We also need to create the control objects unconditionally, yes.

michal-josef-spacek, 2023-04-25:
@rurban Thanks, it's working