LibreDWG/libredwg #713: Wrong POLYLINE and VERTEX parsing in preR13

michal-josef-spacek, 2023-04-27:
The actual situation with parsing of POLYLINE and VERTEX entities in preR13 is this:
* In preR13 is one entity for all POLYLINE and VERTEX types. This means, that the DWG file contains a flag for the (un)presence of variables in the entity and an optional flag (default without this flag is POLYLINE_2D, VERTEX_2D) which means which entity type is present (3D, MESH, PFACE)
* We have naive parsing (polyline - https://github.com/LibreDWG/libredwg/blob/master/src/decode.c#L6614, vertex - https://github.com/LibreDWG/libredwg/blob/master/src/decode.c#L6630) based on entity variables which is not valid for detection.
* There are issues like the entity is presented as POLYLINE_3D, but it's POLYLINE_MESH. Same with VERTEX.

I created [PR](https://github.com/LibreDWG/libredwg/pull/711) for a fix of the polyline, but the issue is with the naivety of parsing.

What about this?
Possibilities:
* Add better detection in the actual block - this means adding parsing of entity flag, which is after elevation, handle, and other common entity things, and will create detection based on this flag.
* Add code for common parsing of preR13 POLYLINE and VERTEX entities - different that POLYLINE_3D (and others) in dwg.spec. And we do some convert to concrete entities.

Example of failing POLYLINE MESH:
[PFACE.DXF.gz](https://github.com/LibreDWG/libredwg/files/11341290/PFACE.DXF.gz)
[PFACE.DWG.gz](https://github.com/LibreDWG/libredwg/files/11341291/PFACE.DWG.gz)

There are other failing situations.

rurban, 2023-05-02:
The r11 VERTEX type check is wrong. I'll change it to use the previous polyline flag

michal-josef-spacek, 2023-05-02:
The same is for the main POLYLINE entity.

michal-josef-spacek, 2023-05-04:
I created PR for fix: https://github.com/LibreDWG/libredwg/pull/719

rurban, 2023-05-06:
Your PR works fine, but is ugly. Maybe create POLYLINE_R11 and VERTEX_R11 specs, and use these private methods then?

michal-josef-spacek, 2023-05-06:
> Your PR works fine, but is ugly. 

I know.

> Maybe create POLYLINE_R11 and VERTEX_R11 specs, and use these private methods then?

I think that the best way is to create POLYLINE_R11 and VERTEX_R11 specs in dwg.spec for read/write.

The question is, in which structures will these specifications be stored?
To this special (like Dwg_Entity_POLYLINE_MESH) or do we need to create common R11 structures?

rurban, 2023-05-06:
I would say to _R11 structs, and then copy the fields over.

rurban, 2023-05-07:
Merged the PR

closed 2023-05-08 by rurban