LibreDWG/libredwg #715: dwgread infinite loop in DWG file generated by libredwg

michal-josef-spacek, 2023-04-27:
There is an infinite loop in the dwgread on file which I generated by the libredwg.

File:
[test.dwg.gz](https://github.com/LibreDWG/libredwg/files/11341524/test.dwg.gz)

rurban, 2023-04-27:
Yes, had the same

rurban, 2023-05-02:
mesh vertex support seems to be missing still. it would be nice to get the json, scr, add or original dwg for this.

michal-josef-spacek, 2023-05-02:
@rurban I don't know which one was. 

Main point of this ticket was infinite loop, but I understand, it's related to bad parsing.

Issue with polylines is described in https://github.com/LibreDWG/libredwg/issues/71

michal-josef-spacek, 2023-05-04:
In [0.12.5.5504](https://github.com/LibreDWG/libredwg/releases/tag/0.12.5.5504) is still present on my instalation. 

We can move to https://github.com/LibreDWG/libredwg/issues/713 and fix this issue, it's probably related.

closed 2023-05-04 by michal-josef-spacek