LibreDWG/libredwg #716: Segfault in dwgadd with layer.color=7

michal-josef-spacek, 2023-04-27:
I have dwgadd code:
```
layer "foo"
layer.color=7
```

Command ```dwgadd -v9 --as r11 -o foo.dwg ex2.dwgadd``` exits with ```Segmentation fault```
It's same for all versions which I tested.

michal-josef-spacek, 2023-04-27:
@rurban I will look to it more.

I believe that is some regression in setting of layer.color in preR11.

rurban, 2023-04-27:
I rather think it's the new r11 handle resolver (r11_idx)

michal-josef-spacek, 2023-04-28:
There are many issues, which wasn't here before.
Code:
```
#include <dwg.h>
#include <dwg_api.h>
#include <string.h>
#include <stdio.h>

#define DWG_LOGLEVEL DWG_LOGLEVEL_ALL

/* Main. */
int
main(int argc, char **argv) {
        Dwg_Data *dwg = NULL;
        Dwg_Object *mspace;
        Dwg_Object_BLOCK_HEADER *hdr;
        dwg_point_3d pt1 = {0, 0, 50};
        dwg_point_3d pt2 = {100.0, 100.0, 100.0};
        int opts = 9;
        static int tracelevel = 9;
        char *filename = "out.dwg";
        int ret;

        memset(&dwg, 0, sizeof(dwg));

        dwg = dwg_new_Document(
                R_1_4,
                0, /* metric/iso */
                tracelevel
        );

        dwg->opts = opts;

        mspace = dwg_model_space_object(dwg);
        hdr = mspace->tio.object->tio.BLOCK_HEADER;

        dwg_add_LINE(hdr, &pt1, &pt2);

        ret = dwg_write_file(filename, dwg);
        if (ret) {
                printf("Write status: %d\n", ret);
                return 1;
        }

        /* Free memory. */
        dwg_free(dwg);

        return 0;
}
```

Created DWG file with:
```
...
CLAYER: -1 [H(RSd) 8]
^^^^^^^^^^^^^^^^^^^^^^
CECOLOR.index: 256 [RS 62]
layer_colors[0]: 0 [RS]
...
entities: (0x202-0x226 (1), size 36)
==========================================
owned by BLOCK *MODEL_SPACE (1f)
@offset: 0x0
type: 1 [RS]
Add entity LINE [8] Decode entity LINE
===========================
Entity number: 8, Type: 1, Addr: 203
layer: -1 [H(RSd) 8]
^^^^^^^^^^^^^^^^^^^^^
object_map{22} = 8
=> handle: (0.1.22)
start: (0.000000, 0.000000) [2RD 10]
end: (100.000000, 100.000000) [2RD 11]
[add handleref (3.1.22) abs:22] BLOCK_HEADER "*MODEL_SPACE".entities[0] = (3.1.22) abs:22 [H]
[add handleref (12.1.3) abs:1F] ownerhandle: [1, 0, 1F]
==========================================
entities: end
...
```

Default layer index from AutoCAD is 1.

rurban, 2023-04-30:
in the smoke/r11-json branch I've fixed that

michal-josef-spacek, 2023-05-01:
ok, segfault is fixed.

closed 2023-05-02 by rurban