LibreDWG/libredwg #717: Failed to install with python3.10: Could not link test program to Python

DaveDixon, 2023-04-30:
Very similar to #521, except that installing python3.10-dev doesn't help - it's already installed. I tried ./configure LIBS="-L/usr/non-standard-path/python/lib" with every possible path to python 3.10 or 3.8 and every library in /usr/lib/python3.x,  and /usr/local/lib/python3.x, and /usr/include/pythonx3.x.

configure: error: 
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LIBS environment variable.
  Example: ./configure LIBS="-L/usr/non-standard-path/python/lib"
  ============================================================================
   ERROR!
   You probably have to install the development version of the Python package
   for your distribution.  The exact name of this package varies among them.
  ============================================================================

config.log doesn't seem to throw an error, per se, but it also seems to think python is in /usr/local/include/python2.7, irrespective of which version I specify in -L and even though the current version is 3.10.

This is Ubuntu 22.04

rurban, 2023-05-02:
We are using the standard autoconf-archive ax_python_devel.m4 which is very often broken. 
Probably ubuntu broke it again. the python people have no idea how get their paths.

I'm using Ubuntu jammy where it works fine.

DaveDixon, 2023-05-02:
Thanks for the reply @rurban! I tried using the PYTHON_VERSION arg to configure - it was still looking for Python2.7. Finally, I tried PYTHON=3.10 and it configure finished without an error. Make, make install, and make check all seemed to complete successfully, so I'll close this. Thanks again!

closed 2023-05-02 by DaveDixon