LibreDWG/libredwg #718: output_SVG  function can't  get the correct result

as010101, 2023-05-04:
when I run these code
`int main(int argc, int ** argv) {

    if (argc != 3) {
        std::cerr << "Usage: " << argv[0] << " input.dwg output.svg" << std::endl;
        return 1;
    }
	int error = 0;
	Dwg_Data dwg;
	memset(&dwg, 0, sizeof(Dwg_Data));
    g_dwg.opts = 3;
	// (iconv_t cd, const char* * inbuf, size_t *inbytesleft, char* * outbuf, size_t *outbytesleft);
	//libiconv(nullptr, nullptr, nullptr, nullptr, nullptr );
	dwg_read_file("example_2018.dwg", &dwg);

	if (!error) {
        output_SVG(&dwg);
return 0;
}
`
I get these output on terminal   [https://github.com/as010101/as010101.github.io/blob/master/img/test.svg](url)
Is there any problem with the use of `output_SVG` , How can I get the svg file?

as010101, 2023-05-04:
> @as010101 You could attach SVG file to this issue. Gzip SVG file and attach it. Link is not working.

I updated.

michal-josef-spacek, 2023-05-04:
**You are using ```output_SVG()``` but this is not something from API.**

There is command line tool ```dwg2SVG``` which could convert your DWG file to SVG and you could look to implementation.
https://github.com/LibreDWG/libredwg/blob/master/programs/dwg2SVG.c

as010101, 2023-05-04:
> **You are using `output_SVG()` but this is not something from API.**
> 
> There is command line tool `dwg2SVG` which could convert your DWG file to SVG and you could look to implementation. https://github.com/LibreDWG/libredwg/blob/master/programs/dwg2SVG.c

Yea. Indeed I have seen the file. `output_SVG` is not api,  the `dwg2SVG.c` file use  'output_svg' convert dwg to svg, I also copy  `output_SVG` function and some relative functions, but cant get correct svg file

michal-josef-spacek, 2023-05-04:
@as010101 I converted your example by actual ```dwg2SVG``` and result is:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:xlink="http://www.w3.org/1999/xlink"
   version="1.1" baseProfile="basic"
   width="100%" height="100%"
   viewBox="-50.321044 -85.321334 270.030212 280.023651">
        <g id="symbol-50" >
                <!-- *Paper_Space -->
        </g>
        <!-- *Model_Space -->
        <!-- circle-44 -->
        <circle id="dwg-object-44" cx="250.002307" cy="170.006558" r="20.000000"
              style="fill:none;stroke:black;stroke-width:0.1px" />
        <text id="dwg-object-45" x="0.002307" y="60.006558" font-family="Courier" font-size="5.000000" fill="black">Jen teksto simpla, cxu ne?</text>
        <!-- lwpolyline-46 -->
        <path id="dwg-object-46" d="M 0.002307,280.006558 L 100.002307,280.006558 L 100.002307,140.006558 L 0.002307,140.006558 Z"
              style="fill:none;stroke:black;stroke-width:0.1px" />
        <!-- line-47 -->
        <path id="dwg-object-47" d="M 150.002307,100.006558 L 250.002307,100.006558"
              style="fill:none;stroke:green;stroke-width:0.1px" />
        <!-- line-48 -->
        <path id="dwg-object-48" d="M 250.002307,0.006558 L 250.002307,100.006558"
              style="fill:none;stroke:black;stroke-width:0.1px" />
        <!-- line-49 -->
        <path id="dwg-object-49" d="M 150.002307,100.006558 L 250.002307,0.006558"
              style="fill:none;stroke:black;stroke-width:0.1px" />
        <defs>
        <g id="symbol-55" >
                <!-- *Paper_Space -->
        </g>
        </defs>
</svg>

This mean, that you have some issue in source code, which I don't see.
```

as010101, 2023-05-04:
> @as010101 I converted your example by actual `dwg2SVG` and result is:
> 
> ```
> <?xml version="1.0" encoding="UTF-8" standalone="no"?>
> <svg
>    xmlns:svg="http://www.w3.org/2000/svg"
>    xmlns="http://www.w3.org/2000/svg"
>    xmlns:xlink="http://www.w3.org/1999/xlink"
>    version="1.1" baseProfile="basic"
>    width="100%" height="100%"
>    viewBox="-50.321044 -85.321334 270.030212 280.023651">
>         <g id="symbol-50" >
>                 
>         </g>
>         
>         
>         <circle id="dwg-object-44" cx="250.002307" cy="170.006558" r="20.000000"
>               style="fill:none;stroke:black;stroke-width:0.1px" />
>         <text id="dwg-object-45" x="0.002307" y="60.006558" font-family="Courier" font-size="5.000000" fill="black">Jen teksto simpla, cxu ne?</text>
>         
>         <path id="dwg-object-46" d="M 0.002307,280.006558 L 100.002307,280.006558 L 100.002307,140.006558 L 0.002307,140.006558 Z"
>               style="fill:none;stroke:black;stroke-width:0.1px" />
>         
>         <path id="dwg-object-47" d="M 150.002307,100.006558 L 250.002307,100.006558"
>               style="fill:none;stroke:green;stroke-width:0.1px" />
>         
>         <path id="dwg-object-48" d="M 250.002307,0.006558 L 250.002307,100.006558"
>               style="fill:none;stroke:black;stroke-width:0.1px" />
>         
>         <path id="dwg-object-49" d="M 150.002307,100.006558 L 250.002307,0.006558"
>               style="fill:none;stroke:black;stroke-width:0.1px" />
>         <defs>
>         <g id="symbol-55" >
>                 
>         </g>
>         </defs>
> </svg>
> 
> This mean, that you have some issue in source code, which I don't see.
> ```

I updated my source code , can you see? please.

as010101, 2023-05-05:
> @as010101 I converted your example by actual `dwg2SVG` and result is:
>
> ```
> <?xml version="1.0" encoding="UTF-8" standalone="no"?>
> <svg
>    xmlns:svg="http://www.w3.org/2000/svg" [...]

thakns. I get correct result by 'sample_2000.dwg' . what if the dwg file is 'example_2018.dwg'? Does this library support converting  from dwg 2018 version   to svg ?  I use 'dwg2SVG' read ''example_2018.dwg' cant get correct result.

michal-josef-spacek, 2023-05-05:
@as010101 
I was looking to example_2018.dwg now. And main issue I have is the size of the image.

**Library is in progress. There are issues in many things.**

If you want to help, you can create a simple DWG + DXF file with an issue, describe the issue, and create a ticket.
There are some SVG-related tickets like https://github.com/LibreDWG/libredwg/issues/523

as010101, 2023-05-05:
> @as010101 I was looking to example_2018.dwg now. And main issue I have is the size of the image.
>
> **Library is in progress. There are issues in many things.**
>
> If you want to help, you can create a [...]

I upload files. and updated in this issue

michal-josef-spacek, 2023-05-05:
> I upload files. and updated in this issue

This is not much helping. This test file is from our repository, it's big and there is no description of the issue.

btw: If you are still updating this issue, we don't see a history of the issue (origin was something in c++, without fix). The main principle of development is to create issues, which developers understand. e.g. I see an issue in conversion of DWG with line to SVG. I will create DWG file with line from 0,0 to 100,100 and generate DXF/SVG file, create ticket about it and describe a situation. It's helping.

michal-josef-spacek, 2023-06-02:
@as010101 Could we close this ticket? This is not about concrete issue.

closed 2023-06-11 by michal-josef-spacek

qwuip789, 2025-07-12:
[image]