LibreDWG/libredwg #722: Add common preR13 POLYLINE and VERTEX entities

michal-josef-spacek, 2023-05-10:
In https://github.com/LibreDWG/libredwg/issues/713 we discussed that PR is ugly and we need common POLYLINE and VERTEX entities.

michal-josef-spacek, 2023-05-10:
I created https://github.com/LibreDWG/libredwg/pull/724 with:
In this PR are:

* Comment updates
* New preR13 POLYLINE and VERTEX structures and API
* preR13 decoding and encoding of DWG.

michal-josef-spacek, 2023-05-10:
@rurban What should be next? Do we need conversion to special entities like POLYLINE_2D in decoding/encoding? Or in different place?

rurban, 2023-05-11:
Hmm, still ugly

michal-josef-spacek, 2023-05-11:
@rurban Do we need some update in dwg_api.c for this new POLYLINE_R11 entity?

rurban, 2023-05-11:
Sure, see my fixups

rurban, 2023-05-12:
Or just use common POLYLINE_R11 and VERTEX_R11 macros in the dwg.spec.
This way no need for extra structs and seperation.

michal-josef-spacek, 2023-05-12:
I am working on it. I will prepare a new PR for it. 
I was fixing DXF yesterday.

rurban, 2023-06-03:
I'll add the other branch soon, with a COMMON_POLYLINE_R11 and VERTEX macro in the spec.

rurban, 2023-06-05:
So now we have now alternative implementations.
* The elegant one with an extra POLYLINE_R11 entity, which has the missing problem of up- downconverting across r11.
* The simple one, with common entity fields, including all the r11 fields, which are not relevant to the special entity in question. We could skip them with compile-time branches (macros).

michal-josef-spacek, 2023-06-05:
Thank you for the implementation of the second one.

I wrote this first one because I thought that you don't want to have all variables in structure. I tend to implementation with the simplest structures personally, but conversions are not easy too. Conversions are only in s few use cases too.

rurban, 2023-06-05:
It's not so important, more convenience.

I"m still fixing other bugs with r11 plines and vertices, but it"s ready soon. #754