LibreDWG/libredwg #727: Wrong setting of header codepage in dwgadd in preR13

michal-josef-spacek, 2023-05-14:
I have dwgadd code:
```
HEADER.codepage=28
point (0.0 0.0 0.0)
```

```dwgadd -v9 --as r11 -o out.dwg code.dwgadd```
generates DWG file with:
```
...
codepage: 30 [RS 0]
...
```

I think that code in src/header_variables_r11.spec:
```
...
  ENCODER {
    FIELD_VALUE (codepage) = dwg->header.codepage;
  }
  FIELD_RS (codepage, 0);
  DECODER {
    dwg->header.codepage = FIELD_VALUE (codepage);
  }
...
```
is bad

michal-josef-spacek, 2023-05-14:
btw: point entity is not created in actual master

rurban, 2023-08-11:
Well, HEADER.codepage=28 sets header_vars.CODEPAGE, and is then overwritten by header.codepage. Looks like this needs a dwg_dynapi_header_set_value special case.

closed 2023-08-11 by rurban

michal-josef-spacek, 2023-08-15:
Great, thank you.