LibreDWG/libredwg #729: error code 68

berry0511, 2023-05-19:
src file:  [huafei.zip](https://github.com/LibreDWG/libredwg/files/11514535/huafei.zip)

code:
    auto dwg_ = new Dwg_Data;
    int success;
    dwg_->num_objects = 0;
    success = dwg_read_file(file_path_.string().c_str(), dwg_);

return val:
    success = 68
dwg_errstrings shows: UNHANDLEDCLASS VALUEOUTOFBOUNDS 

version: latest master

berry0511, 2023-05-19:
further more:
using dwgread try to open the file shows:
[image]

rurban, 2023-05-19:
Yes, and?
We cannot read all classes and never will without the accompaning ARX sources

berry0511, 2023-05-19:
> 

> Yes, and? We cannot read all classes and never will without the accompaning ARX sources

what is the ARX sources?

rurban, 2023-05-21:
> Yes, and? We cannot read all classes and never will without the
> accompaning ARX sources
>
> what is the ARX sources?
DWG allows external dynamic objects, which are compiled to ARX. Many
companies build their own objects with ARX, and they dont give away how
their objects are stored.

rurban, 2023-06-02:
More worrying is 

Warning: checksum: 0x3008130e (calculated) mismatch
Warning: Invalid sections: 0 != numgaps: 0 + numsections: 189

closed 2023-06-02 by rurban