LibreDWG/libredwg #741: Wrong dwgadd parsing of float/int in HEADER variables

michal-josef-spacek, 2023-05-25:
When I have input "HEADER.VIEWSIZE = 1.2345"

Actual code
```
...
      else if (2 == SSCANF_S (p, "HEADER." FMT_NAME " = %d\n", s1 SZ, &i1))
        dwg_dynapi_header_set_value (dwg, s1, &i1, 0);
      else if (2 == SSCANF_S (p, "HEADER." FMT_NAME " = %lf\n", s1 SZ, &f1))
        dwg_dynapi_header_set_value (dwg, s1, &f1, 0);
...
```
couldn't detect float. 
The code matches the first rule, the value of VIEWSIZE will be 1, and dwg_dynapi_header_set_value() set nothing.

rurban, 2023-05-25:
Oh, we have to switch it. Thanks

rurban, 2023-05-30:
Fixed

michal-josef-spacek, 2023-05-30:
I don't think that's working. In this time float is working, but int not. 

```dwgadd -v9 --as r1.4 -o test.dwg test.dwgadd```
```
HEADER.LUNITS=3
arc (0.0 1.0 0.0) 0.5 0.0 3.0
```

rurban, 2023-05-31:
we have to check all sizes and types when poking the value into the structure via dynapi.

rurban, 2023-06-01:
Fixed with 7c120ed96d237e8a82369131a391b30d678144e1

closed 2023-06-01 by rurban