LibreDWG/libredwg #745: Regression: Conversions from DWG to JSON in case of FIELD_TFv

michal-josef-spacek, 2023-05-31:
Example:
[FOO.DWG.gz](https://github.com/LibreDWG/libredwg/files/11618479/FOO.DWG.gz)

```dwgread -o FOO.json FOO.DWG```

There is bad output:
```
...
    {
      "object": "VIEW",
      "index": 26,
      "type": 61,
      "handle": [0, 0, 26],
      "size": 91,
      "flag": 0,
      "name": "WIDE\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
      "VIEWSIZE": 1.10429447852761,
      "VIEWCTR": [ 3.18915918684925, 6.99386503067485 ],
      "view_width": 6.62682428436208,
      "VIEWDIR": [ 0.0, 0.0, 1.0 ],
      "flag_3d": 0
    }
...
```

Interesting is, that it's one of name fields in DWG file.
As I understand, this is regression from last month

rurban, 2023-05-31:
The \u0 are actually valid Unicode to fill the buffer properly, but then I optimized the ending slack away. Not really a bug.

michal-josef-spacek, 2023-06-01:
Commit above is working

michal-josef-spacek, 2023-06-02:
Hm, I'll leave it open. There is some weird issue in AC1001 in MENU, I don't know what is it for now.

closed 2023-06-02 by rurban

michal-josef-spacek, 2023-06-02:
Ok, I will investigate and create another bug. Thanks.