LibreDWG/libredwg #747: Issue in setting MENU header variable

michal-josef-spacek, 2023-06-02:
I have example:
```
#include <dwg.h>
#include <dwg_api.h>
#include <string.h>
#include <stdio.h>

#define DWG_LOGLEVEL DWG_LOGLEVEL_ALL

/* Main. */
int
main(int argc, char **argv) {
        Dwg_Data *dwg = NULL;
        int opts = 9;
        static int tracelevel = 9;
        char *filename = "out.dwg";
        int ret;
        char text[120] = "foo";

        memset(&dwg, 0, sizeof(dwg));

        dwg = dwg_new_Document(
                R_2000,
                0, /* metric/iso */
                tracelevel
        );

        dwg->opts = opts;

        dwg_dynapi_header_set_value (dwg, "MENU", text, 1);

        ret = dwg_write_file(filename, dwg);
        if (ret) {
                printf("Write status: %d\n", ret);
                return 1;
        }

        /* Free memory. */
        dwg_free(dwg);

        return 0;
}
```
Compile: ```gcc -std=c99 -Wall `pkg-config --cflags libredwg` -o foo foo.c `pkg-config --libs libredwg```

Gets Segmentation fault

michal-josef-spacek, 2023-06-02:
The root cause is fix in https://github.com/LibreDWG/libredwg/pull/748
After the fix, this segfault is caused.

michal-josef-spacek, 2023-06-02:
Construct:
```
...
char *text = "foo";
...
dwg_dynapi_header_set_value (dwg, "MENU", &text, 1);
...
```
is working

rurban, 2023-06-02:
yes, I know. just fixing it

rurban, 2023-06-02:
Fixed

closed 2023-06-02 by rurban