LibreDWG/libredwg #749: dwgadd and setting of entity layer

michal-josef-spacek, 2023-06-02:
In doc is construct like:
```
line (0 0 0) (100 0 100)
line.layer = "1"
```

But output is:
```
ERROR: dwg_dynapi_entity_set_value: Invalid LINE field layer
```
in ```dwgadd -v9 --as r11 -o line.dwg foo.dwgadd```

rurban, 2023-06-02:
you cannot reference a layer, which does not exist yet. so the error is correct

michal-josef-spacek, 2023-06-02:
> you cannot reference a layer, which does not exist yet. so the error is correct

ok, that's a good point. Thanks

But when I have:
```
layer "foo"
line (0 0 0) (100 0 100)
line.layer = "foo"
```
The layer was created and error is same

rurban, 2023-06-02:
this does not work yet, as the layer must be a handle, not a name.
maybe we should accept table names also

rurban, 2023-06-02:
This is up to r12, and should work already