LibreDWG/libredwg #758: Improve flags_r11 setting in preR13 encoding

michal-josef-spacek, 2023-06-07:
I am going to improve flags_r11 setting.
I created PR https://github.com/LibreDWG/libredwg/pull/757 which:
* Describe these numbers precisely
* Added flags_r11 from the structure if is defined

One think is TODO, which I am not able to set now:
* Add ```dwg->header.section[id].flags_r11 = __FROM_CONTROL__``` in https://github.com/LibreDWG/libredwg/blob/master/src/in_json.c#L3264 It helps conversions from JSON to DWG and better ability to test conversion from DWG to JSON and back.

rurban, 2023-06-08:
This comment `TODO: maybe add a missing CONTROL object here. GH #453` is something else.
We do store all the CONTROL objects plus its flags_r11 in the json already.

closed 2023-06-08 by rurban

michal-josef-spacek, 2023-06-08:
@rurban In JSON, yes, but conversion from JSON to DWG is not working (e.g. in the case of flags_r11)