LibreDWG/libredwg #761: Question: What about default LTYPE values like BYLAYER, BYBLOCK, CONTINUOUS in preR13?

michal-josef-spacek, 2023-06-10:
In postR13 is in the header handle for these default LTYPE values.

But in preR13 it quietly does not find a handle for these types.
```
...
          if (vars->LTYPE_BYLAYER)
            return vars->LTYPE_BYLAYER;
...
```
in https://github.com/LibreDWG/libredwg/blob/master/src/dwg.c#L2580-L2581

In the case of CONTINUOUS, there is the process of creating this LTYPE before this setting. Then there is the possibility to set vars->LTYPE_CONTINUOUS after creating.

rurban, 2023-06-12:
bylayer and byblock are magic indices, if I remember correctly, without actual table records nor handles. this must be specialized.

rurban, 2023-08-17:
not setting HEADER.LTYPE_CONTINUOUS is just a missing opimization. It still has to walk the entries, and compare the names then.

rurban, 2023-08-17:
Is this it?

michal-josef-spacek, 2023-08-18:
I am trying to remember what was the root cause of this question.

I think conversion from DXF to DWG. I will look to it.