LibreDWG/libredwg #767: convert error in dwg2dxf

weikenxq, 2023-06-25:
Hello,

I'm trying to convert dwg to dxf using dwg2dxf, there are some errors and the dxf file is not generated.

Are there any way to solve the errors? 

Thank you.

### environment
macOS big sur 11.5.2 , Apple clang version 12.0.5 (clang-1205.0.22.9),GNU Make 3.81, libredwg latest [commit](https://github.com/LibreDWG/libredwg/commit/a0636e94f35cfc140cccfb026dabb1de2742977f)

compile with

./autogen.sh && ./configure --disable-bindings && make && sudo make install

### some information
programs/dwg2dxf -o ~/Downloads/6151acdedf1bc01e5a5847868d8bdbbee139b29c.dwg.dxf  ~/Downloads/6151acdedf1bc01e5a5847868d8bdbbee139b29c.dwg -y
Reading DWG file /Downloads/6151acdedf1bc01e5a5847868d8bdbbee139b29c.dwg
Warning: 
@0x6834 => start 0x7b5d
Warning: 
@0x7e61 => start 0x8508
ERROR: DWG_SENTINEL_R11_VIEW_BEGIN not found at 34056
ERROR: Failed to decode file: /Downloads/6151acdedf1bc01e5a5847868d8bdbbee139b29c.dwg 0x100

origin dwg file open with AutoCAD:
[image]

### attachment
[6151acdedf1bc01e5a5847868d8bdbbee139b29c.dwg.zip](https://github.com/LibreDWG/libredwg/files/11859001/6151acdedf1bc01e5a5847868d8bdbbee139b29c.dwg.zip)

michal-josef-spacek, 2023-06-25:
@weikenxq This DWG file with "errors". Question is how to deal with this situations.

btw: I regen this file with R11:
[F.DWG.gz](https://github.com/LibreDWG/libredwg/files/11860478/F.DWG.gz)
[F.DXF.gz](https://github.com/LibreDWG/libredwg/files/11860479/F.DXF.gz)

michal-josef-spacek, 2023-06-25:
@weikenxq You could try ```dwg2dxf``` on the my file, but there are errors still. We need to fix step by step.

weikenxq, 2023-06-26:
@mi

> @weikenxq You could try `dwg2dxf` on the my file, but there are errors still. We need to fix step by step.

Thanks for reply. I try dwg2dxf on the regen file, I can get the converted result and the result looks ok. But this is just one sample, there are lots of files having the same problem with dwg2dxf. Is it possible to deal with this errors when use dwg2dxf?

weikenxq, 2023-06-27:
> @weikenxq You could try `dwg2dxf` on the my file, but there are errors still. We need to fix step by step.

Thanks for reply. I try dwg2dxf on the regen file, I can get the converted result and the result looks ok. But this is just one sample, there are lots of files having the same problem with dwg2dxf. Is it possible to deal with this errors when use dwg2dxf?

michal-josef-spacek, 2023-06-27:
@weikenxq I am looking to preR13 things periodically. I am going roundtripping via JSON and DXF. And I see issues.

rurban, 2023-06-27:
Exactly those two warnings are the problem:'
The table addresses are wrong.

Warning:
@0x6834 => start 0x7b5d
Warning:
@0x7e61 => start 0x8508

Maybe check against the natural address for the sentinel, when the table.address fails.

leshasoft, 2023-06-27:
The file has a non-standard structure.
But ODA converter converts without problems.

[qq.zip](https://github.com/LibreDWG/libredwg/files/11881249/qq.zip)

michal-josef-spacek, 2023-06-27:
(original file)
This mean, there are missing sections which we need to deduce.
And tables:
```
ptr table VIEW [6]: (0x8518-0x8518)
VIEW.size: 153 [RS]
VIEW.number: 0 [RS]
VIEW.flags_r11: 0x0 [RS]
VIEW.address: 34072 [RL]

ptr table UCS [7]: (0x8518-0x8518)
UCS.size: 109 [RS]
UCS.number: 0 [RS]
UCS.flags_r11: 0x0 [RS]
UCS.address: 34072 [RL]

ptr table VPORT [8]: (0x8518-0x8518)
VPORT.size: 253 [RS]
VPORT.number: 0 [RS]
VPORT.flags_r11: 0x0 [RS]
VPORT.address: 34072 [RL]

ptr table APPID [9]: (0x8528-0x854d)
APPID.size: 37 [RS]
APPID.number: 1 [RS]
APPID.flags_r11: 0x0 [RS]
APPID.address: 34088 [RL]
```
are bad.

michal-josef-spacek, 2023-06-27:
@weikenxq 
1) This previous two comments are about what I am talking. There are serious issues in DWG file structure. We could deduce and somehow fix.
2) Second thing is real dwg2dxf conversion, which has own issues but in DXF values (precision, order in DXF, some differences between autocad release dxf output and other issues)

weikenxq, 2023-06-28:
> @weikenxq
>
> 1. This previous two comments are about what I am talking. There are serious issues in DWG file structure. We could deduce and somehow fix.
> 2. Second thing is real dwg2dxf conversion, [...]

@michal-josef-spacek Thanks. When should I expect this enhancement implemented as a part of this library?

rurban, 2023-06-28:
I'm just busy with with oda compat and DXF roundtrips. michal hopefully

michal-josef-spacek, 2023-06-30:
ad 1) ok, I will prepare a patch that could detect bad section and skip it.

rurban, 2023-07-07:
We don't need to skip bad sections, as we can search for them by the sentinel's

closed 2026-11-04 by rurban