LibreDWG/libredwg #769: Failed to decode file

weikenxq, 2023-07-04:
Hello,

I try to convert dwg to dxf using dwg2dxf, got error as below:

dwg2dxf -o ~/Downloads/64a08d92d5b6a0c965b9442fb0b48fb7389b9cec.dwg.dxf /Downloads/64a08d92d5b6a0c965b9442fb0b48fb7389b9cec.dwg
Reading DWG file /Downloads/64a08d92d5b6a0c965b9442fb0b48fb7389b9cec.dwg
ERROR: Failed to decode file: /Users/weiken/Downloads/64a08d92d5b6a0c965b9442fb0b48fb7389b9cec.dwg 0x100

READ ERROR 0x100

origin dwg file open with AutoCAD:
[image]

### attachment
[64a08d92d5b6a0c965b9442fb0b48fb7389b9cec.dwg.zip](https://github.com/LibreDWG/libredwg/files/11949891/64a08d92d5b6a0c965b9442fb0b48fb7389b9cec.dwg.zip)

rurban, 2023-07-06:
This is a `section_map_address: 0x0 [RLL 0]`, which we cannot deal with yet. No idea how to find the real section_map_address now. ODA can read it, but their specs don't define that case.

rurban, 2023-07-07:
Fixed, thanks for the testcase!

closed 2023-07-07 by rurban