LibreDWG/libredwg #771: How to use this? dlls missing

Aabha-cell, 2023-07-05:
I downloaded the libredwg-0.12.5.5885-win64.zip and extracted it. I'm trying to run the dwg2svg2.exe. I'm getting a popup saying libredwg-0.dll was not found. How do I get this? Do I need to download anything else?
Please help and guide on how to run the exe files on windows system</div>

rurban, 2023-07-06:
No, this is the content of  libredwg-0.12.5.5885-win64.zip, and the dll is there.
[image]

zefreestijl, 2024-03-18:
I got the same problem recently,
in Windows System Properties -> Environment Variables, 
edit and add a new Path of the directories where .dll file is located.

rurban, 2024-03-19:
On windows you just need to keep the dll's in the very same dir as the exe's. Other than that you might need a depends.exe -like tool, which reports which dll's are missing. We cannot help with that.

closed 2024-03-19 by rurban