LibreDWG/libredwg #775: DWG to JSON and back strings_area issue

michal-josef-spacek, 2023-07-19:
In https://github.com/LibreDWG/libredwg/issues/346 I described roundtripping issue in conversion between JSON and DWG.

In the conversion from DWG to JSON is used:
```
DWG_TABLE (LTYPE)
...
FIELD_BINARY (strings_area, 256, 0);
```
in dwg.spec
The result of the conversion of "" is 512 character string with zeros.

In the conversion from JSON to DWG is used:
```
in_hex2bin() ...
```
in in_dxf.c
The result of the conversion of 512 character string with zeros is "0".

I am curious when we have in_hex2bin without tests, and there is a copy of this function in unit-testing/add_test.c.

I think that could be fine to have one function and tests with it. + Fix of this issue.

michal-josef-spacek, 2023-07-19:
btw: I don't know if usage of ```in_hex2bin()``` is fine in this use case.
I was looking to ```FIELD_BINARY``` in in_json(), but this is not used in this case. This is not intuitive too

michal-josef-spacek, 2023-07-19:
@rurban Could we move this ```in_hex2bin()``` to some common c file? 
I am proposing bits.c :-)

rurban, 2023-07-19:
Sure. It's used by both importers

michal-josef-spacek, 2023-07-19:
PR for move: https://github.com/LibreDWG/libredwg/pull/776

michal-josef-spacek, 2023-07-21:
@rurban Thanks for the tests

michal-josef-spacek, 2023-07-21:
Hm, this issue is invalid. Error is before this, probably in saving of numdashes.

michal-josef-spacek, 2023-07-21:
Heh funny bug.

michal-josef-spacek, 2023-07-21:
PR for fix: https://github.com/LibreDWG/libredwg/pull/778

rurban, 2023-07-24:
Merged

closed 2023-07-24 by rurban