LibreDWG/libredwg #785: DWG to JSON (and back) and CMC issue

michal-josef-spacek, 2023-07-24:
In https://github.com/LibreDWG/libredwg/issues/346 is discussed issue:
```
1206,1207c1206
<             "index": 256,
<             "rgb": "c3000007"
---
>             "rgb": "c0000000"
```

There is the conversion of fill_color (CMC) defined by (AC1015):
```
rowstyles[rcount1].fill_color.index: 256 [CMC.BS 63]
rowstyles[rcount1].fill_color.rgb: 0xc3000007 [CMC.BL 421]
fill_color.method: 0xc3 Truecolor
rowstyles[rcount1].fill_color.flag: 0x0 [CMC.RC]
```
to JSON, where is:
```
          "fill_color": {
            "index": 256,
            "rgb": "c3000007"
          },
```
This means that rgb color doesn't exist in index table → index=256
See:
```
bit_read_CMC()
{
...
color->index = dwg_find_color_index (color->rgb);
...
}
```
and
```
EXPORT BITCODE_BS
dwg_find_color_index (BITCODE_BL rgb)
{       
  Dwg_RGB_Palette pal;
  rgb &= 0x00ffffff;
  pal.r = (rgb >> 16) & 0xFF;
  pal.g = (rgb >> 8) & 0xFF; 
  pal.b = rgb & 0xFF;
  // linear search is good enough for 256. the palette is unsorted.
  for (BITCODE_BS i = 0; i < 256; i++)
    {     
      if (memcmp (&pal, &rgb_palette[i], 3) == 0)
        return i;
    }   
  return 256;
}
```

When I am converting this JSON back to DWG, result will be:
```
rowstyles[rcount1].text_color.flag: 0x0 [CMC.RC]
rowstyles[rcount1].fill_color.index: 0 [CMC.BS 63]
rowstyles[rcount1].fill_color.rgb: 0xc0000000 [CMC.BL 421]
fill_color.method: 0xc0 ByLayer
rowstyles[rcount1].fill_color.flag: 0x0 [CMC.RC]
```
The problem with this is, that was called:
```
      if (dat->from_version >= R_2004)
        bit_downconvert_CMC (dat, color);
```
in bit_write_CMC().

michal-josef-spacek, 2023-07-24:
Root cause of this issue is in:
```
// from old palette to r2004+ truecolor (FIXME)
void
bit_upconvert_CMC (Bit_Chain *dat, Dwg_Color *restrict color)
{
  if (dat->version >= R_2004 && dat->from_version < R_2004)
    {
      if (!color->method)
        color->method = 0xc3;
      if (color->index == 256)
        color->method = 0xc0;
      else if (color->index == 0)
        color->method = 0xc1;

      color->rgb = color->method << 0x18;
      if (color->method == 0xc3)
        color->rgb |= dwg_rgb_palette_index (color->index);
    }
}
```
Where detect index = 256 and set ```color->method = 0xc0;```

michal-josef-spacek, 2023-07-24:
I don't know what is exactly bad.
I believe that's no fine automatically convert to R_2004 if it's AC1015.
And probably is bad:
```
bit_write_CMC (....)
...
      if (dat->from_version < R_2004)
        bit_upconvert_CMC (dat, color);
...
```
which call this conversion.

michal-josef-spacek, 2023-07-24:
Same bad call of ```bit_upconvert_CMC()``` is in use case:
```
5927c5924
<             "rgb": "c8000000"
---
>             "rgb": "c1000000"
```
only different if:
```
      else if (color->index == 0)
        color->method = 0xc1;
```

michal-josef-spacek, 2023-07-24:
@rurban What's the fix?
Is it missing ```dat->from_version = R_2004``` before convert?

rurban, 2023-07-31:
One wrong case is that TABLE classes (and other) with color fields do keep the method and rgb slots with r2000. So extend write_CMC with the obj to check if downconvert is needed

rurban, 2023-07-31:
No, CMC: downconvert not for classes is wrong. In the spec this is a CMTC field already, but this logic is missing for encode.

michal-josef-spacek, 2023-07-31:
FYI: When I tested this issue, the code:
```
---------------------------------- src/bits.c ----------------------------------
index 1439ead4e..7782a8edf 100644
@@ -3369,8 +3369,8 @@ bit_write_CMC (Bit_Chain *dat, Bit_Chain *str_dat, Dwg_Color *restrict color)
 {
   if (dat->version >= R_2004) // truecolor
     {
-      if (dat->from_version < R_2004)
-        bit_upconvert_CMC (dat, color);
+//      if (dat->from_version < R_2004)
+//        bit_upconvert_CMC (dat, color);
       bit_write_BS (dat, 0); // index override
       bit_write_BL (dat, color->rgb);
       if (!color->method && color->rgb & 0xFF000000)
```
was fixed my issue precisely :-) I don't think that's fine

rurban, 2023-07-31:
Can you check if smoke/cmc fixes that?

michal-josef-spacek, 2023-07-31:
Seems ok, thanks

closed 2023-08-01 by rurban