LibreDWG/libredwg #788: Issues in Template section

michal-josef-spacek, 2023-07-31:
When I am investigating https://github.com/LibreDWG/libredwg/issues/346 I found issues in Template section

Support of templates is from R14
Template section consists from description and MEASUREMENT fields:
There are two types of Template section:
1. AC1013 .. AC1018
description is RS for text size and RC for each byte
MEASURENMENT is RS
3. AC1021 .. ?
description is RS for text size and RS for each byte
MEASURENMENT is RS

Issues:
a. In AC1013 .. AC1015 isn't description decoding. And bad comment "(0 English/256 Metric)"
b. No write support for description
c. There is no write support from JSON file to DWG for template description
d. How to support text in encoding? For each versions.

michal-josef-spacek, 2023-07-31:
I prepared a partial fix in https://github.com/LibreDWG/libredwg/pull/787

Questions:
- There is _T16, but for the second version of template section, we need a different helper function. 
- What about the text in some encoding?

Fixing of c) after these fixes.

michal-josef-spacek, 2023-07-31:
Example:
[test.dwt.gz](https://github.com/LibreDWG/libredwg/files/12213522/test.dwt.gz)

After fix in PR:
```
=======> MEASUREMENT 4 (start)  :     3165
         MEASUREMENT 4 (end)    :     3174
description: "..zek" [T16 0]
MEASUREMENT: 0 [RS 0]
-> HEADER.MEASUREMENT: 0 (0 English/1 Metric)
```

Text is the Czech word ```řízek```

rurban, 2023-07-31:
Oh, the encoding is still wrong.

michal-josef-spacek, 2023-07-31:
description (AC1013 .. AC1018) is T16
description (AC1021 ) is TU16

@rurban Could we store both versions in one variable?

rurban, 2023-07-31:
Sure, as with all other texts

michal-josef-spacek, 2023-07-31:
I don't know how. As I see we don't have situation T16 with TU16.

When I change:
```
diff --git a/include/dwg.h b/include/dwg.h
index 4fc546ac7..dab585fcc 100644
--- a/include/dwg.h
+++ b/include/dwg.h
@@ -9554,7 +9554,7 @@ typedef struct _dwg_objfreespace
 
 typedef struct _dwg_template
 {
-  BITCODE_T16 description;
+  BITCODE_T description;
   BITCODE_RS MEASUREMENT;
 } Dwg_Template;
```

and
```
diff --git a/src/template.spec b/src/template.spec
index ca19625f5..0e80d7ff7 100644
--- a/src/template.spec
+++ b/src/template.spec
@@ -18,5 +18,10 @@
 
 #include "spec.h"
 
-FIELD_T16 (description, 0);
+UNTIL (R_2004) {
+  FIELD_T16 (description, 0);
+}
+SINCE (R_2007) {
+  FIELD_TU16 (description, 0);
+}
 FIELD_RS (MEASUREMENT, 0); // copied to header_vars
```

There is an error:
```
In file included from decode.c:74:
template.spec: In function ‘template_private’:
dec_macros.h:568:15: error: assignment to ‘BITCODE_TV’ {aka ‘char *’} from incompatible pointer type ‘BITCODE_TU’ {aka ‘short unsigned int *’} [-Werror=incompatible-pointer-types]
  568 |     _obj->nam = bit_read_TU16 (dat);                                          \
      |               ^
template.spec:25:3: note: in expansion of macro ‘FIELD_TU16’
   25 |   FIELD_TU16 (description, 0);
      |   ^~~~~~~~~~
^Cmake[4]: *** [Makefile:730: decode.lo] Chyba 1
```

rurban, 2023-08-01:
Fixed

michal-josef-spacek, 2023-08-01:
This is not working in the case of TU16.

When I update the code:
include:
BITCODE_TU description_tu;
src:
FIELD_TU16 (description_tu, 0);
Then it's working right.

This means, that the cast from TU16 to T16 isn't working. Sure?

rurban, 2023-08-02:
Yes, we'd need to up/downconvert also.

michal-josef-spacek, 2023-08-03:
@rurban Thanks for code.
dwgread for versions >= R_2007 is fine. Text in some encoding is printed as utf8.

Only thing is output encoding for <R_2007.
Example (codepage 28, ```řízek``` in description):
[czech.dwt.gz](https://github.com/LibreDWG/libredwg/files/12248930/czech.dwt.gz)
Actual output:
```
=======> MEASUREMENT 4 (start)  :    19473
         MEASUREMENT 4 (end)    :    19482
description: "..zek" [T16 0]
MEASUREMENT: 0 [RS 0]
-> HEADER.MEASUREMENT: 0 (0 English/1 Metric)
```

rurban, 2023-08-03:
Now fixed

closed 2023-08-03 by rurban

michal-josef-spacek, 2023-08-15:
@rurban Do you plan UTF-8 output in logging?

Example:
I have an example with the Czech name of the template in AC1014. Conversion to JSON is fine, output is in UTF-8. But dwgread prints binary data in log :-)

rurban, 2023-08-15:
Currently only asian. Old dwg's  with old codepages stay untranslated so far. I have no opinion on this.

Technically it should check the system locale and convert to this long term

michal-josef-spacek, 2023-08-16:
ok, thank you for your answer. I think that could be interesting to unify to output.

rurban, 2023-08-16:
it should also add the INSANE hex output with non-standard codepages on -v5

michal-josef-spacek, 2023-08-16:
Yes, and some hex output in the case when is e.g. ASCII terminal. :-)