LibreDWG/libredwg #789: AcDb:Handles start address

michal-josef-spacek, 2023-07-31:
When I am investigating https://github.com/LibreDWG/libredwg/issues/346 I found an issue with AcDb:Handles section

Process:
- DWG file from https://github.com/LibreDWG/libredwg/issues/346
```
section[2].number:     2 [RC] AcDb:Handles
section[2].address: 163035 [RL]
section[2].size:     678 [RL]
```
- Convert to JSON
- Convert back to DWG
```
section[2].number:     2 [RC] AcDb:Handles
section[2].address: 163037 [RL]
section[2].size:    1167 [RL]
```

Issues:
1. First two sections are ok, but the third section start address is different.
2. And items inside are different.

michal-josef-spacek, 2023-07-31:
Example issue:
```
Next object: 20 Handleoff: 0x1 [UMC]
==========================================
Object number: 20/14, Size: 63 [MS], Type: 73 [BS]
Add object MLINESTYLE [20] Decode object MLINESTYLE
bitsize: 470 [RL] @3.2
 Hdlsize: 0x22,
handle: 0.1.18 [H 5]

num_eed: 0
num_reactors: 1 [BL 0]
ownerhandle: (8.0.0) abs:23 [H 330]
reactors[0]: (4.1.17) abs:23 [H* 330]
xdicobjhandle: (3.0.0) abs:0 [H 360]
name: "STANDARD" [T 2]
description: "" [T 0]
flag: 0 [BS 70]
fill_color.index: 256 [CMC.BS 62]
start_angle: 1.5708 [BD 51]  90º
end_angle: 1.5708 [BD 52]  90º
num_lines: 2 [RC 71]
lines[0].offset: 0.5 [BD 49]
lines[rcount1].color.index: 256 [CMC.BS 62]
lines[rcount1].lt_index: 32767 [BSd 6]

lines[1].offset: -0.5 [BD 49]
lines[rcount1].color.index: 256 [CMC.BS 62]
lines[rcount1].lt_index: 32767 [BSd 6]

crc: 9850 [RSx]
```

vs (after conversion from JSON)
```
Next object: 20 Handleoff: 0x1 [UMC]
==========================================
Object number: 20/14, Size: 63 [MS], Type: 73 [BS]
Add object MLINESTYLE [20] Decode object MLINESTYLE
bitsize: 470 [RL] @3.2
 Hdlsize: 0x22,
handle: 0.1.18 [H 5]

num_eed: 0
num_reactors: 1 [BL 0]
ownerhandle: (8.0.0) abs:23 [H 330]
reactors[0]: (4.1.17) abs:23 [H* 330]
xdicobjhandle: (3.0.0) abs:0 [H 360]
name: "STANDARD" [T 2]
description: "" [T 0]
flag: 0 [BS 70]
fill_color.index: 256 [CMC.BS 62]
start_angle: 1.5708 [BD 51]  90º
end_angle: 1.5708 [BD 52]  90º
num_lines: 2 [RC 71]
lines[0].offset: 0.5 [BD 49]
lines[rcount1].color.index: 256 [CMC.BS 62]
lines[rcount1].lt_index: 32767 [BSd 6]

lines[1].offset: -0.5 [BD 49]
lines[rcount1].color.index: 256 [CMC.BS 62]
lines[rcount1].lt_index: 32767 [BSd 6]

crc: 7DFD [RSx]
```

Decoding output is same, but inside is somethig different (different crc).

michal-josef-spacek, 2023-07-31:
Other failing objects are: LAYOUT, MATERIAL, SECTIONVIEWSTYLE and XRECORD

rurban, 2023-08-01:
The different crc's are due to different float encodings, and we dont want to store floats as hex. Not a bug, just mildly annoying.

closed 2023-08-01 by michal-josef-spacek