LibreDWG/libredwg #792: 2NDHEADER in JSON

michal-josef-spacek, 2023-07-31:
When I am investigating https://github.com/LibreDWG/libredwg/issues/346 I found that 2NDHEADER isn't in JSON.
When I convert JSON to DWG, the DWG is without 2NDHEADER.

What about adding 2NDHEADER to JSON?
Or do we need to fix the conversion from JSON to DWG by adding 2NDHEADER?

rurban, 2023-08-01:
We can add this section, yes. For JSON it's redundant, but for roundtrips it would help

rurban, 2023-08-02:
Now enabled. Maybe we shouldn't write it in release mode.

michal-josef-spacek, 2023-08-02:
Thanks, I will test it

michal-josef-spacek, 2023-08-03:
@rurban 
Interesting issue is with conversion to JSON and back
DWG (dwgread -v3)
```
=======> Second Header 3 (start):   163713
...
version: AC1015 [TFF 12]
...
```
Conversions to JSON, output:
```
...
  "SecondHeader": {
    "version": "AC1015\u0000\u0000\u0000\u0000\u0000\u000f",
...
```
Conversion to DWG, dwgread output:
```
=======> Second Header 3 (start):   164204
...
version: AC1015 [TFF 12]
...
```
Conversion to JSON, againg from previous:
```
...
  "SecondHeader": {
    "version": "AC1015",
...
```

When I am looking to sources:
src/in_json.c:
```
FIELD_TFF (version, 12, 0)
```
src/out_json.c:
```
FIELD_TF (version, 12, 0);
```

michal-josef-spacek, 2023-08-03:
Second issue:
When I convert from DWG to JSON and back.
In Second header in DWG file is:
```
section[i].nr: 0x0 [RC 0]
section[i].address: 0x3723 [BLx 0]
section[i].size: 512 [BL 0]
section[i].nr: 0x1 [RC 0]
section[i].address: 0x3923 [BLx 0]
section[i].size: 1048 [BL 0]
section[i].nr: 0x2 [RC 0]
section[i].address: 0x27cdd [BLx 0]
section[i].size: 1167 [BL 0]
section[i].nr: 0x3 [RC 0]
section[i].address: 0x2816c [BLx 0]
section[i].size: 0 [BL 0]
section[i].nr: 0x0 [RC 0]
^^^^^^^^^^^^^^^^^^^ → We need 0x4
section[i].address: 0x0 [BLx 0]
section[i].size: 0 [BL 0]
section[i].nr: 0x5 [RC 0]
section[i].address: 0x61 [BLx 0]
section[i].size: 123 [BL 0]
```

rurban, 2023-08-11:
Fixed both

michal-josef-spacek, 2023-08-15:
The first one is ok. Thanks.

The second one is not ok.
The issue is the bad creating of template record in the second header.

Input JSON file:
[test.json.gz](https://github.com/LibreDWG/libredwg/files/12345288/test.json.gz)

Command to convert: ```dwgwrite test.json```
Output in dwg file is:
```
sections[4].nr: 0 [RCd 0]
sections[4].address: 0 [BL 0]
sections[4].size: 0 [BL 0]
```
We need something like:
```
sections[4].nr: 4 [RCd 0]
sections[4].address: 164349 [BL 0]
sections[4].size: 4 [BL 0]
```

rurban, 2023-08-16:
Because we write the template after the 2ndheader, and we calc. the dat->byte (address) only then.

so we might need to patch in both section addresses. the first and the 2nd. or calc the template address and size before writing the 2ndheader

closed 2023-08-16 by rurban

michal-josef-spacek, 2023-08-16:
@rurban Thanks