LibreDWG/libredwg #797: libredwg-0.12.4.4364-win64 Failed to parse AC1018 DWG drawing with error code 0xd41

lym-ID, 2023-08-04:
error:
[image]
bug file:
[dwg.zip](https://github.com/LibreDWG/libredwg/files/12256366/dwg.zip)

rurban, 2023-08-04:
Yes, starting with 
Page: 2106     size: 121220 address: 0x280000041b000000 info: 0x608680
Page: 469763075 (a) size:   429 address: 0xa0b0000083b

it gets corrupted. The section page reader logic is not right yet.

lym-ID, 2023-08-04:
Hello author, we are currently blocked by this bug. Our team is working on a function to publish maps, and your libreDWG is the core of this function. May I ask how long it will take to fix this bug?

rurban, 2023-08-04:
This is a long-standing libredwg bug, for several years already. It is caused by deleting a large number of entities, which leads to gaps in an internal tree, which is somehow optimized into the section pages. I have no idea when we can fix it.
See GH #439, #272 , #144