LibreDWG/libredwg #801: Build error

prokoudine, 2023-08-13:
Building 0ad71990 on Ubuntu 23.04 with GCC 12.3.0 fails:

```
escape.c: In function ‘htmlescape’:
escape.c:41:25: error: pointer ‘dest’ used after ‘realloc’ [-Werror=use-after-free]
   41 |       const int off = d - dest;
      |                       ~~^~~~~~
escape.c:45:25: note: call to ‘realloc’ here
   45 |           dest = (char*)realloc (dest, len);
      |                         ^~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

Please let me know what other information I should provide to help you figure this out.

risicle, 2023-08-19:
Building 0.12.5.6173 on nixos we're still getting:

```
escape.c: In function 'htmlescape':
escape.c:41:25: error: pointer 'dest' may be used after 'realloc' [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wuse-after-free-Werror=use-after-free8;;]
   41 |       const int off = d - dest;
      |                       ~~^~~~~~
escape.c:44:22: note: call to 'realloc' here
   44 |           d = (char*)realloc (dest, len + 10);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~
```

rurban, 2023-08-20:
That's unfortunately a gcc-12 compiler bug. It's missing the dest = d line 47 in its' var-tracking. Let me rewrite it.

rurban, 2023-08-21:
Fixed now

closed 2023-08-21 by rurban