LibreDWG/libredwg #805: Don't decode AC1014 thumbnail

michal-josef-spacek, 2023-08-17:
Example file:
[line.dwg.gz](https://github.com/LibreDWG/libredwg/files/12373403/line.dwg.gz)

Decoding output:
```
=======> Thumbnail:       3165
```

Detection of thumbnail:
```
  00000c5d 3080 DWG_SENTINEL_THUMBNAIL_BEGIN [..16]
  000050f8 3081 DWG_SENTINEL_THUMBNAIL_END [..16]
```

Begin address is right, but searching for a second sentinel doesn't work.

michal-josef-spacek, 2023-08-17:
Found in investigating https://github.com/LibreDWG/libredwg/issues/346

michal-josef-spacek, 2023-08-17:
Issue is from AC1012 to AC1014

rurban, 2023-08-18:
yes, a long standing dwgbmp issue

michal-josef-spacek, 2023-08-18:
In this case code:
```
if (bit_search_sentinel (dat, dwg_sentinel (DWG_SENTINEL_THUMBNAIL_END)))
        {
          BITCODE_RL bmpsize;
          LOG_TRACE ("         Thumbnail (end): %4zu\n", dat->byte)
```
(https://github.com/LibreDWG/libredwg/blob/master/src/decode.c#L433)

doesn't find sentinel, but sentinel is in place (last 16 bytes of data block). 

The issue is with the find on the end of the block.
We need to change (src/bits.c:bit_search_sentinel):
```
-  for (i = 0; i < dat->size - 16; i++)
+  for (i = 0; i <= dat->size - 16; i++)
```

michal-josef-spacek, 2023-08-18:
https://github.com/LibreDWG/libredwg/pull/807

closed 2023-08-18 by rurban