LibreDWG/libredwg #809: Issue in encoding 3DLINE in preR13

michal-josef-spacek, 2023-08-18:
There is an issue in the encoding of 3DLINE in pre13.
Test via dwgadd.

cat test.dwgadd
``` 
line (1 2 3) (4 5 6)
```

Command:
```dwgadd -v9 --as r9 -o test.dwg test.dwgadd```

Output:
```
Add entity _3DLINE [8] Decode entity _3DLINE
===========================
Entity number: 8, Type: 21, Addr: 3ef
flag_r11: 0x0 [RC 0]
size: 56 [RS]
layer: -1 [H(RSd) 8]
opts_r11: 0x3 [RSx 0]
          HAS_Z_FIRST(0x1) HAS_Z_SECOND(0x2)
object_map{22} = 8
=> handle: (0.1.22)
start: (0, 1, 2) [3RD 10]
end: (3, 4, 5) [3RD 11]
```

michal-josef-spacek, 2023-08-18:
I was looking at ```dwg_add_LINE``` API function, and there is no possibility of an issue.

btw: Isn't better add ```dwg_add_3DLINE()```  low level function too?

The only place where could be the issue is dwg.spec and definition of 3DLINE.

michal-josef-spacek, 2023-08-18:
When I add:
```
 /* (none/21) R2.4-R10 only */
 DWG_ENTITY (_3DLINE)
   VERSIONS (R_2_4, R_9c1) {
+    LOG_TRACE ("XXX (%f %f %f)\n", _obj->start.x, _obj->start.y, _obj->start.z);
     if (R11OPTS (1)) {
       FIELD_3RD (start, 10);
```

In encoding is output:
```
Entity LINE, number: 8, Addr: 1007 (0x3ef)
type: 21 [RC]
Encode entity _3DLINE
flag_r11: 0x0 [RC 0] @2.0
size: 0 [RS] @4.0
layer: -1 [RSd 8] @6.0
opts_r11: 0x3 [RSx 0] @8.0
XXX (0.000000 1.000000 2.000000)
start: (0.000000, 1.000000, 2.000000) [3RD 10] @32.0
end: (3.000000, 4.000000, 5.000000) [3RD 11] @56.0
-size: 56 [RL] (@1011.0)
```

rurban, 2023-08-19:
Lgtm. I dont think we should add an extra 3DLINE adder, rather handle it transparently, as of now

michal-josef-spacek, 2023-08-19:
ad extra 3DLINE function) The idea was here because in the R10 is a possibility of both and we don't allow to set explicitly now. I think that's better for testing. This is a minor thing.

ad issue)
The result should be:
```
start: (1, 2, 3) [3RD 10]
end: (4, 5, 6) [3RD 11]
```

And I don't understand, why is moved to 0, 1, 2 and 3, 4, 5.
When we processed in dwgadd it's ok, when we processed in dwg_add_LINE it's ok because normal LINE is functional. 
This is something related to 3DLINE.

Another idea is something bad in src/common_entity_data.spec, but I don't see it.

michal-josef-spacek, 2023-08-19:
@rurban Isn't issue with ```API_ADD_ENTITY (LINE);``` ?
https://github.com/LibreDWG/libredwg/blob/master/src/dwg_api.c#L23932

There is preparation of LINE entity and this is rewritten to 3DLINE in some situations.

michal-josef-spacek, 2023-08-24:
There is a question whether not to implement low level function for 3DLINE.  (I am for, improve possibility of better testing)
Add low level construct of 3dline to dwgadd.  (I am for, improve possibility of better testing)

And add process for move line to 3dline in DWG versions which haven't line with 3d support.
And probably other place, but not in low level functions.