LibreDWG/libredwg #818: What about BLOCK_HEADER in DWG->JSON conversion in preR13?

michal-josef-spacek, 2023-08-24:
Actual situation:

Example (Blank AC2.10 DWG file)
[BLANK.DWG.gz](https://github.com/LibreDWG/libredwg/files/12431679/BLANK.DWG.gz)
After conversion to JSON, there is BLOCK_HEADER in the form:
```
...
{
}
...
```
Because code (in src/out_json.c):
```
...
     case DWG_TYPE_BLOCK_HEADER:
      if (dat->version <= R_12
          && strEQc (obj->tio.object->tio.BLOCK_HEADER->name, "*MODEL_SPACE"))
        {
          LOG_TRACE ("Skip *MODEL_SPACE\n");
          return 0;
       }
...
```

Which generates an ERROR log in conversion from JSON to DWG:
```
ERROR: Required (null).type missing, skipped
```
Because JSON ```{}```.

I think that is bad.
Some possibilities:
1) Remove {} from JSON, this means no error log.
2) Remove the code above and generate JSON code like:
```
    {
      "object": "BLOCK_HEADER",
      "index": 5,
      "type": 49,
      "handle": [0, 1, 31],
      "size": 0,
      "flag": 0,
      "name": "*MODEL_SPACE",
      "block_offset_r11": 4294967295,
      "unknown_r11": 0
    },
```
Which seems somehow right.
The result after conversion from JSON to DWG is fine.

@ruban What do you think?

rurban, 2023-09-05:
I forgot why I skipped it there. It can be skipped in encode easier.
Variant 2 please

michal-josef-spacek, 2023-09-05:
I need to look. There were more changes in https://github.com/LibreDWG/libredwg/commit/e5f27c184438fadfd3e0fec89eaa2de4b622a13a