LibreDWG/libredwg #823: change dwgwrite version defaults from r2000 to asis

rurban, 2023-08-31:
new version defaults, don't write r2000, but old versions asis
    
dwg readers would prefer r2000 or higher, oda can only read from r9 on,
but for the sake of sanity and less up/downconversions don't upconvert preR13 to r2000 per default.
only <r1.4 upconvert to r1.4, and >=r2004 down to r2000.
This way we have much less up/downconverting changes.

michal-josef-spacek, 2023-08-31:
There is only one issue with AC1.40, which is a different format.
Like:
```
EXTENTS,1
0.000000,9.862348,0.000000,6.354656
LIMITS,1
0.000000,12.000000,0.000000,9.000000
BASE,1
0.000000,0.000000
DWGVIEW,1
6.000000,4.634818,9.269636
MODERES,1
0,1.000000
MODEGRID,1
0,0.000000
MODEORTHO,1
1
MODEFILL,1
1
TXTSIZE,1
0.200000
TRACEWID,1
0.050000
LAYER,1
1
LAYERC,1
0,15,255,255,255,255,255,255,255,255,255,15,255,255,255,255
255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
LINE,1
3.255061,0.932794,9.862348,5.246964
LINE,1
9.862348,5.246964,4.663968,6.354656
LINE,1
4.663968,6.354656,4.663968,2.798381
```

Other versions are okay, we need to fix issues in them. This change will allow these errors to be corrected.

michal-josef-spacek, 2023-08-31:
I am talking about DXF :-)

rurban, 2023-08-31:
I forgot if we can properly round trip <r1.4 json already (DXF not for sure due to the extra format). but I"ve merged dwgwrite to keep r2.x-r13

closed 2023-08-31 by rurban