LibreDWG/libredwg #828: JSON file generated by libredwg is not valid - duplicate keys

michal-josef-spacek, 2023-08-31:
I have JSON output like:
```
    {
      "entity": "BLOCK",
      "index": 23,
      "type": 4,
      "handle": [0, 1, 33],
      "size": 29,
      "bitsize": 187,
      "_subclass": "AcDbEntity",
      "layer": [5, 1, 16, 16],
      "prev_entity": [4, 0, 0, 0],
      "next_entity": [4, 0, 0, 0],
      "preview_exists": 0,
      "entmode": 1,
      "isbylayerlt": 1,
      "nolinks": 0,
      "color": 256,
      "ltype_scale": 1.0,
      "invisible": 0,
      "_subclass": "AcDbBlockBegin",
      "name": "*PAPER_SPACE"
    },
```

There are two '_subclass' keys. This is not possible in JSON.

rurban, 2023-08-31:
Well, that depends. duplicate keys are permitted with certain readers.

https://stackoverflow.com/questions/21832701/does-json-syntax-allow-duplicate-keys-in-an-object

michal-josef-spacek, 2023-08-31:
@rurban Ok, interesting.

I could go with it, thanks for information.

closed 2023-08-31 by rurban