LibreDWG/libredwg #835: DXF2DWG occurs  this error:Invalid DXF group code :6400

gf0126, 2023-09-05:
Hi,I have a dxf file used dxf2dwg.exe to convert it to dwg,but it ocuurs error as follows:

 D:\code\dwg\libredwg>dxf2dwg.exe 456.dxf
Reading DXF file 456.dxf
ERROR: Invalid DXF group code: 6400
ERROR: Failed to decode DXF file: 456.dxf

READ ERROR 0x800 456.dxf

and I give the dxf file ,please give me an answer!
[456.zip](https://github.com/LibreDWG/libredwg/files/12522602/456.zip)

rurban, 2023-09-07:
This is a binary DXF. They are not yet supported. Normally DXFB import fails much earlier

gf0126, 2023-09-08:
Hello, I have changed the dxf. Then use the dxf2dwg.exe,It occurs error:Invalid DXF Code 294 for DIMSTYLE.
I give the dxf file, please give me a reply! Thank you!
[456(0909).zip](https://github.com/LibreDWG/libredwg/files/12558335/456.0909.zip)

rurban, 2024-02-05:
This has a DXF code of 294 with a DIMSTYLE, which accoring to my specs is invalid. Esp. with r2000. Maybe it's valid for r2010, because there exists a DIMTXTDIRECTION with code 295, but I have no docs for that, which boolean field that could be.

rurban, 2024-02-05:
Fixed with 0.13.1, at least the ASCII DXF part

rurban, 2024-02-05:
And the binary part seems to be corrupt. 0x1900 as code is invalid, and therefore rejected

closed 2024-02-05 by rurban