LibreDWG/libredwg #836: ERROR 0x540

turtle0x1, 2023-09-06:
Hey, long time no support request - great work on the library! :wave: 

Ive got a DWG i'm trying to read to JSON and get the following errors;

```
$ ../programs/dwgread -O json -o ./dwgread-DR.json dr_test.dwg -v3
Warning: checksum: 0x54100440 (calculated) CRC mismatch 0x13afbc0-0x13afbd0

Warning: Invalid sections: 0 != numgaps: 10 + numsections: 2255
ERROR: Invalid opcode 0x8 in input stream at pos 20558221
Warning: Failed to find section_info[0] with type 1
ERROR: Failed to read compressed Header section
ERROR: Failed to read uncompressed SummaryInfo section
Warning: Failed to find section_info[0] with type 3
ERROR: Failed to read compressed Classes section
Warning: Failed to find section_info[0] with type 7
ERROR: Failed to read compressed AcDbObjects section
Warning: Failed to find section_info[0] with type 2
ERROR: Failed to read uncompressed AuxHeader section
ERROR: Failed to read uncompressed Preview section
ERROR: Failed to read uncompressed AppInfo section
ERROR: Template section not found

ERROR: Failed to decode file: dr_test.dwg 0x540

ERROR 0x540
```

In this particular case I could ask the designer to tidy up (anonymize) the drawing and submit it, but over the next few days ill be running this on a few hundred thousand DWG's, which I suspect will contain more errors.

At that many drawings we're abit beyond "tidy up and submit" for bug fixes, is there any log/information that would be useful from parsing "so many" DWGs at once? Maybe some kind of aggregate statistics about common errors?

The file dates go back years but I suspect the earliest file format is whatever was shipped with ACAD 2011, the latest being whatever is being shipped with ACAD now.

rurban, 2023-09-24:
I suspect this error appears when somewhen is deleting a lot of entities. Related to the section page map bugs (deleted pages in the tree)

rurban, 2024-02-06:
Cannot help much without the dr_test.dwg

closed 2024-02-06 by rurban