LibreDWG/libredwg #841: 0 tests passed

turtle0x1, 2023-09-07:
Ultra - "nit pickty" - "All 0 tests pasts" is both confusing and needless - can It be > 0 or not show atol?

```
====================
All 252 tests passed
====================
make[4]: Leaving directory '/home/libredwg/libredwg/test/unit-testing'
make[3]: Leaving directory '/home/libredwg/libredwg/test/unit-testing'
make[3]: Entering directory '/home/libredwg/libredwg/test'
make[3]: Nothing to be done for 'check-am'.
make[3]: Leaving directory '/home/libredwg/libredwg/test'
make[2]: Leaving directory '/home/libredwg/libredwg/test'
Making check in doc
make[2]: Entering directory '/home/libredwg/libredwg/doc'
echo make regen-dynapi might be needed
make regen-dynapi might be needed
make[2]: Leaving directory '/home/libredwg/libredwg/doc'
make[2]: Entering directory '/home/libredwg/libredwg'
make  check-TESTS
make[3]: Entering directory '/home/libredwg/libredwg'
==================
All 0 tests passed

```

rurban, 2023-09-07:
No idea. This is an autotools thing. Google for it

qwuip789, 2023-09-20:
1070 warnings and 2 errors generated.
make[3]: *** [Makefile:733: decode.lo] Error 1
make[3]: Leaving directory '/home/qw/libredwg/src'
make[2]: *** [Makefile:752: install-recursive] Error 1
make[2]: Leaving directory '/home/qw/libredwg/src'
make[1]: *** [Makefile:704: install-recursive] Error 1
make[1]: Leaving directory '/home/qw/libredwg'
make: *** [Makefile:1093: install] Error 2

closed 2023-09-20 by rurban