LibreDWG/libredwg #843: LibreDWG Compiling an Android library with NDK on Ubuntu is not successful.

qwuip789, 2023-09-20:
How to proceed step by step, please provide a solution.

qwuip789, 2023-09-20:
decode.c:6013:5: warning: cast from 'BITCODE_TV' (aka 'char *') to 'BITCODE_TU' (aka 'unsigned short *') increases required alignment from 1 to 2 [-Wcast-align]
    COMMON_ENTITY_DIMENSION
    ^~~~~~~~~~~~~~~~~~~~~~~
./dwg.spec:1626:9: note: expanded from macro 'COMMON_ENTITY_DIMENSION'
        FIELD_T (user_text, 1);                                               \
        ^~~~~~~~~~~~~~~~~~~~~~
./dec_macros.h:578:33: note: expanded from macro 'FIELD_T'
            LOG_TRACE_TU (#nam, (BITCODE_TU)FIELD_VALUE (nam), dxf)           \
            ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./logging.h:145:44: note: expanded from macro 'LOG_TRACE_TU'
      LOG_TEXT_UNICODE (TRACE, (BITCODE_TU)wstr)                              \
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
./logging.h:158:51: note: expanded from macro 'LOG_TEXT_UNICODE'
      if (DWG_LOGLEVEL >= DWG_LOGLEVEL_##level && wstr)                       \
                                                  ^~~~
decode.c:6013:5: warning: cast from 'BITCODE_TV' (aka 'char *') to 'BITCODE_TU' (aka 'unsigned short *') increases required alignment from 1 to 2 [-Wcast-align]
    COMMON_ENTITY_DIMENSION
    ^~~~~~~~~~~~~~~~~~~~~~~
./dwg.spec:1626:9: note: expanded from macro 'COMMON_ENTITY_DIMENSION'
        FIELD_T (user_text, 1);                                               \
        ^~~~~~~~~~~~~~~~~~~~~~
./dec_macros.h:578:33: note: expanded from macro 'FIELD_T'
            LOG_TRACE_TU (#nam, (BITCODE_TU)FIELD_VALUE (nam), dxf)           \
            ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./logging.h:145:44: note: expanded from macro 'LOG_TRACE_TU'
      LOG_TEXT_UNICODE (TRACE, (BITCODE_TU)wstr)                              \
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
./logging.h:160:60: note: expanded from macro 'LOG_TEXT_UNICODE'
          ATTRIBUTE_ALIGNED(2) char *_u8 = bit_convert_TU (wstr);             \
                                                           ^~~~
1070 warnings and 2 errors generated.
make[3]: *** [Makefile:730: decode.lo] Error 1
make[3]: Leaving directory '/home/qw/libredwg/src'
make[2]: *** [Makefile:749: install-recursive] Error 1
make[2]: Leaving directory '/home/qw/libredwg/src'
make[1]: *** [Makefile:702: install-recursive] Error 1
make[1]: Leaving directory '/home/qw/libredwg'
make: *** [Makefile:1091: install] Error 2

rurban, 2023-09-20:
-Wcast-align warnings are not fatal

closed 2023-09-20 by rurban

qwuip789, 2023-09-20:
> -Wcast-align 警告不是致命的

 [Makefile：730：decode.lo]错误1
[Makefile:1091: install] 错误 2

qwuip789, 2023-09-20:
[Makefile:730: decode.lo] Error 1
[Makefile:1091: install] Error 2