LibreDWG/libredwg #844: How can I use the LibreDWG library on an Android smartphone?

qwuip789, 2023-09-20:
FATAL EXCEPTION: main
Process: com.example.cadviwer, PID: 8631
  java.lang.UnsatisfiedLinkError: dlopen failed: library "libredwg.so.0" not found: needed by /data/app/~~ymWA7_c29qFzOomORNeq8w==/com.example.cadviwer-TsQ4zoJndvEAVx0JlHtfkg==/base

qwuip789, 2023-09-20:
./configure --host=arm-linux-androideabi --prefix=/home/qw/arm

qwuip789, 2023-09-20:
export ANDROID_NDK=/home/qw/android-ndk-r21d
     export TOOLCHAIN=/home/qw/cross_tool/AndroidToolChain
     export NDK_CROSS=/home/qw/cross_tool/AndroidToolChain/bin
     PATH=$PATH:$ANDROID_NDK:$NDK_CROSS
sudo apt-get install autoconf automake libtool
/home/qw/android-ndk-r21d/build/tools/make-standalone-toolchain.sh --platform=android-21 --toolchain=arm-linux-androideabi-4.8 --arch=arm --install-dir=/home/qw/cross_tool/AndroidToolChain

qwuip789, 2023-09-20:
make install file not generated

rurban, 2023-09-21:
We don"t give advice how to develop or use basic libraries, esp. on unsupported platforms which we cannot test. But maybe a user with basic knowledge of the Android SDK can help you. Looks like the LD_LIBRARY_PATH is missing

closed 2023-09-21 by rurban