LibreDWG/libredwg #848: Clarification on Open Bug Report

coolkingcole, 2023-09-26:
Hello Maintainers,

I have a bug report open on huntr.dev, available [here](https://huntr.dev/bounties/8ccb62e1-d25a-45af-87fd-b9e4f0a5087b/), which has been pending for some time. I am keen to assist in resolving this promptly and am willing to close it if deemed out of scope or invalid. Your guidance on this would be appreciated.

Thanks in advance for your time and consideration.

rurban, 2023-09-26:
You need to file bug reports here, or at savannah. We certainly don't look at other sites

coolkingcole, 2023-09-27:
So that these reports on the unsupported platform can be closed swiftly and in one place, I will add the 2 other reports here for tracking.

Report was validated by a maintainer and no cve assigned:
https://huntr.dev/bounties/305219e6-ba85-4310-b036-f0e00910fe72/

Second report:
https://huntr.dev/bounties/c30300ff-4da4-4759-800f-e4340b04f019/

I can close these on request.

rurban, 2023-09-27:
I will check later. Am on holidays for a week

closed 2023-10-07 by rurban

coolkingcole, 2023-10-08:
Can you assign a CVE if deemed noteworthy? If not, I can close the reports on huntr.dev.