LibreDWG/libredwg #849: Attributes not shown in output file

kovacevski, 2023-09-26:
Hi,
I successfully convert the dwg file to geojson but all the attributes are missing.
For example
In my dwg file I have something like this
[image]
and in my geojson I cannot see this properties. 
`{
      "type": "Feature",
      "id": "3DA",
      "properties": {
        "SubClasses": "AcDbEntity : AcDbBlockReference",
        "Layer": "UnitDetail",
        "Color": 256,
        "name": "UnitDetail",
        "EntityHandle": "3DA"
      },
      "geometry": {
        "type": "Point",
        "coordinates":         [ 368462.7367884437553585, 2682140.4038846166804433, -0.0044009092129904 ]
      }
    }`
Is there any way to extract this attributes as key value pairs?

rurban, 2023-10-18:
As properties, or what? can you point me to a geojson example?

kovacevski, 2024-05-21:
The command that I am running is: dwgread -O geojson
[output_geojson.json](https://github.com/LibreDWG/libredwg/files/15391061/output_geojson.json)
[test.zip](https://github.com/LibreDWG/libredwg/files/15391062/test.zip)

rurban, 2026-11-06:
HATCH fixed with PR #1502 by Garcia Leal

garcialeal, 2026-11-06:
The original request, the block attributes, is in #1504: each ATTRIB of an INSERT or MINSERT is written as a "TAG": "value" property of its feature, an empty value as "". Constant attributes, which are only in the block definition, are not covered yet.

closed 2026-11-07 by rurban