LibreDWG/libredwg #851: Fatal error autocad

turtle0x1, 2023-10-01:
I converted a DWG with an empty model space and one paper space layout to JSON using `dwgread`. I then converted that JSON back to DWG using `dwgwrite`. When trying to open the DWG AutoCAD 2024 fatal errors ("unhandled access violation").

[image]

When doing the `dwgwrite` I get a fair few errors/warnings (just a sample of the "scariest" looking ones);

```
ERROR: Required object or entity key missing
ERROR: Unknown MTEXTOBJECTCONTEXTDATA.num_unknown_bits 488 ignored
ERROR: Unexpected STRING at 1386408 of 1509528 tokens, expected OBJECTS OBJECT. json_OBJECTS:3042
ERROR: Expected JSON STRING
ERROR: Unknown num_dashes field
ERROR: Unknown EVALUATION_GRAPH.unknown_bits ....
ERROR: Unknown subclass field EVAL_Node.node
Warning: Object handle not found 3305242624/0xC5020000
```

Would it be okay to email over the DWG?

Also I see encoding only supports outputting to the the 2000 format, is writing to a later format still on the roadmap?

rurban, 2023-10-02:
> Would it be okay to email over the DWG?

Yes, please

rurban, 2023-10-02:
> Also I see encoding only supports outputting to the the 2000 format, is writing to a later format still on the roadmap?

Sure. But for this period we favored the easier preR13 tasks

turtle0x1, 2023-10-02:
> > Would it be okay to email over the DWG?
> 
> Yes, please

Still at the gmail address?

rurban, 2023-10-02:
any you find

rurban, 2023-10-02:
The producer fails with ../oda gh_851_gh851.json:
  Object of class AcDbDictionary can't be cast to AcDbEntity.

turtle0x1, 2023-10-02:
What could be causing so many dictionary objects?

turtle0x1, 2023-10-04:
The above Autocad still renders attribute tags / mtext within a block on the paperspace as pink for some reason, any ideas? Maybe something I could look at? (I'll need some pointers)

But down to 75 errors on open, 25 removed again, a 43% decrease :tada: (I know this a non-goal, but still nice!)

turtle0x1, 2023-10-05:
This one causes autocad to crash and never load for me

rurban, 2023-10-11:
> This one causes autocad to crash and never load for me

That's bad, because we need that long-term

rurban, 2023-11-29:
Still the same problem

turtle0x1, 2023-11-29:
Still crashes when using the JSON generated by the old version using the command `./programs/dwgwrite -I json -o  out.dwg in.dwg`. Will try regenerating the JSON file.

rurban, 2024-02-05:
Missing implemention of MTEXTATTRIBUTEOBJECTCONTEXTDATA, causing these errors.
Which needs the Embedded Object logic for a SCALE object.

turtle0x1, 2024-02-06:
I think your implying I should retest this? If so I can do that by the end of week if thats alright?

rurban, 2024-02-06:
I've already tested it. It doesn't crash anymore but mtext attributes are still not supported, leading to this error.