LibreDWG/libredwg #856: Possible error in dwg_add_SEQEND (dwg_api.h)

abwen, 2023-10-23:
The type of the blkhdr parameter does not seem correct:

typedef struct _dwg_entity_POINT dwg_ent_generic;
...
 EXPORT Dwg_Entity_SEQEND *
 dwg_add_SEQEND (**_dwg_ent_generic_** *restrict blkhdr) 

As far as I understand, it should be like this:

EXPORT Dwg_Entity_SEQEND *
dwg_add_SEQEND (**_Dwg_Object_BLOCK_HEADER_** *restrict blkhdr)

rurban, 2023-11-24:
Not a bug. Just the name is wrong, the type is correct.
The owner of a SEQEND might by a a POLYLINE or an INSERT. But since we use the generic API_ADD_ENTITY macro with the comment /* globals: dxfname, blkhdr=owner */ it should be clear enough, that the blkhdr in this case (the owner) is not a BLOCK_HEADER type.

rurban, 2023-11-24:
Well, I fixed the docs for this

closed 2023-11-24 by rurban