LibreDWG/libredwg #860: Fix second header block

michal-josef-spacek, 2023-11-03:
This is a tracking issue for activity on the second header block.
I found issues in the case of https://github.com/LibreDWG/libredwg/issues/346

michal-josef-spacek, 2023-11-03:
Examples (all AC1012 DWG files):
* [BLANK.DWG.gz](https://github.com/LibreDWG/libredwg/files/13248284/BLANK.DWG.gz)
* [DEMO.DWG.gz](https://github.com/LibreDWG/libredwg/files/13248285/DEMO.DWG.gz)
* [LINE1.DWG.gz](https://github.com/LibreDWG/libredwg/files/13248286/LINE1.DWG.gz)

BLANK.DWG:
* Blank drawing saved in AutoCAD R13
Actual header
```
This file's version code is: AC1012 (r13)
zero_5: { 0x0 0x0 0x0 0x0 0x0 } [*RC]
is_maint: 0x0 [RC 0]
zero_one_or_three: 0x1 [RC 0]
thumbnail_address: 562 [RL 0]
dwg_version: 0x0 [RC 0]
maint_version: 0x0 [RC 0]
codepage: 12 [RS 0]
CP850

num_sections: 3 [RL]
num_sections => 3
section[0].number:     0 [RC] AcDb:Header
section[0].address:   70 [RL]
section[0].size:     454 [RL]
section[1].number:     1 [RC] AcDb:Classes
section[1].address:  524 [RL]
section[1].size:      38 [RL]
section[2].number:     2 [RC] AcDb:Handles
section[2].address: 3977 [RL]
section[2].size:      79 [RL]
crc: A24C [RSx] from 0-52
         HEADER (end):      70
```
Actual Second header
```
=======> Second Header 3 (start):     4056
size: 90 [RL 0]
address: 4056 [BL 0]
version: "AC1012" [TFF 11 0]
4143313031320000000000
0xf [RC 0]
null_b: { 0 0 0 0 } [*B]
unknown_10: 0x18 [RC 0]
unknown_rc4: { 0x30 0x1 } [*RC]
num_sections: 3 [RC 0]
sections[0].nr: 0 [RCd 0]
sections[0].address: 70 [BL 0]
sections[0].size: 454 [BL 0]
sections[1].nr: 1 [RCd 0]
sections[1].address: 524 [BL 0]
sections[1].size: 38 [BL 0]
sections[2].nr: 2 [RCd 0]
sections[2].address: 3977 [BL 0]
sections[2].size: 79 [BL 0]
num_handles: 14 [BS 0]
handles[0].num_hdl: 1 [RCd 0]
handles[0].nr: 0 [RCd 0]
hdl[0]: 0x23 [RC 0]
[0] HANDSEED: 0.1.23
handles[1].num_hdl: 1 [RCd 0]
handles[1].nr: 1 [RCd 0]
hdl[0]: 0x1 [RC 0]
[1] BLOCK_CONTROL_OBJECT: 0.1.1
handles[2].num_hdl: 1 [RCd 0]
handles[2].nr: 2 [RCd 0]
hdl[0]: 0x2 [RC 0]
[2] LAYER_CONTROL_OBJECT: 0.1.2
handles[3].num_hdl: 1 [RCd 0]
handles[3].nr: 3 [RCd 0]
hdl[0]: 0x3 [RC 0]
[... 259 lines cut ...]
hdl[0]: 0x9 [RC 0]
[8] APPID_CONTROL_OBJECT: 0.1.9
handles[9].num_hdl: 1 [RCd 0]
handles[9].nr: 9 [RCd 0]
hdl[0]: 0xa [RC 0]
[9] DIMSTYLE_CONTROL_OBJECT: 0.1.A
handles[10].num_hdl: 1 [RCd 0]
handles[10].nr: 10 [RCd 0]
hdl[0]: 0xb [RC 0]
[10] VX_CONTROL_OBJECT: 0.1.B
handles[11].num_hdl: 1 [RCd 0]
handles[11].nr: 11 [RCd 0]
hdl[0]: 0xc [RC 0]
[11] DICTIONARY_NAMED_OBJECT: 0.1.C
handles[12].num_hdl: 1 [RCd 0]
handles[12].nr: 12 [RCd 0]
hdl[0]: 0x17 [RC 0]
[12] DICTIONARY_ACAD_MLINESTYLE: 0.1.17
handles[13].num_hdl: 1 [RCd 0]
handles[13].nr: 13 [RCd 0]
hdl[0]: 0xd [RC 0]
[13] DICTIONARY_ACAD_GROUP: 0.1.D
crc: 0xc11d [RSx 0]
         Second Header 3 (end)  :    26159
```

michal-josef-spacek, 2023-11-03:
Issues or things to improve:
1.  We could probably change MAGIC to string with 10 characters + \0 or 11 characters. Actually MAGIC + zero_5
2. Fix nonsense structure
3. We could probably merge dwg_version + maint_version to RS, because in the second header is BS type
An alternative possible write is read_BB and if (result == 0) read_dwg_version && read_maint_version, but this is not right.
4. What about checking CRC in the second header?
5. Fix the same things in the aux header as 2)
6. What about warning "Warning: Only 3 num_sections, but 2ndheader found, extending to 4". There are regular DWG files with 3, 4, and 5 sections.

michal-josef-spacek, 2023-11-03:
First PR: https://github.com/LibreDWG/libredwg/pull/861

michal-josef-spacek, 2023-11-03:
Output after #861:

BLANK.dwg:
```=======> Second Header 3 (start):     4056
size: 90 [RL 0]
address: 4056 [BL 0]
version: "AC1012" [TFF 6 0]
414331303132
zero_5: { 0x0 0x0 0x0 0x0 0x0 } [*RC]
is_maint: 0x0 [RC 0]
zero_one_or_three: 0x1 [RC 0]
dwg_version: 0 [BS 0]
=> header dwg_version: 0x0
=> header maint_version: 0x0
codepage: 12 [RS 0]
num_sections: 3 [BS 0]
...
crc: D5BF [RSx]
 check_CRC 4072-4160 = 88: D5BF == D5BF
```

DEMO.dwg:
```
=======> Second Header 3 (start):   191443
size: 100 [RL 0]
address: 191443 [BL 0]
version: "AC1012" [TFF 6 0]
414331303132
zero_5: { 0x0 0x0 0x0 0x0 0x0 } [*RC]
is_maint: 0x6 [RC 0]
zero_one_or_three: 0x1 [RC 0]
dwg_version: 0 [BS 0]
=> header dwg_version: 0x0
=> header maint_version: 0x0
codepage: 30 [RS 0]
num_sections: 4 [BS 0]
...
crc: A0B3 [RSx]
 check_CRC 191459-191557 = 98: A0B3 == A0B3
```

LINE1.DWG:
```
=======> Second Header 3 (start):    26013
size: 114 [RL 0]
address: 26013 [BL 0]
version: "AC1012" [TFF 6 0]
414331303132
zero_5: { 0x0 0x0 0x0 0x0 0x0 } [*RC]
is_maint: 0x6 [RC 0]
zero_one_or_three: 0x1 [RC 0]
dwg_version: 1559 [BS 0]
=> header dwg_version: 0x17
=> header maint_version: 0x6
codepage: 28 [RS 0]
num_sections: 5 [BS 0]
...
crc: 18DC [RSx]
 check_CRC 26029-26141 = 112: 18DC == 18DC
```

michal-josef-spacek, 2023-11-04:
@rurban 
In my change I moved crc and junk_14 outside of the spec file, then it is not in JSON output (=issue).
Q1: Should be CRC in JSON or not? 
Q2: How to deal with the situation? Move back and add a check after using of spec file?

rurban, 2023-11-04:
More importantly, encode is missing it also. I've added it

rurban, 2023-11-07:
Looks good now. Agreed?

michal-josef-spacek, 2023-11-08:
Yes, it's better.
* Writing support is done.
* dwg to json and back is working
* dwgadd is working
* I looked at the aux header and I don't understand what this means now.

There is an issue with reading in Autocad 2000 still :-/

michal-josef-spacek, 2023-11-08:
@rurban what do you think about merging MAGIC with zeroes? Or stay two fields?

michal-josef-spacek, 2023-11-09:
Another PR for fix of encoding: https://github.com/LibreDWG/libredwg/pull/865

michal-josef-spacek, 2023-11-09:
I removed 9), because the structure seems ok, but real Autocad doesn't read the file.

michal-josef-spacek, 2023-11-11:
Updated PR
After https://github.com/LibreDWG/libredwg/pull/865:
BLANK.DWG:
```
=======> Second Header 3 (start):     4056
before_sentinel: 0x01010e012f011101854b012801f308028902019548019b0201c44401cf4101250120019d410125012001bf0a05d041d9eb000201d0 [53]
size: 90 [RL 0]
address: 4056 [BL 0]
version: "AC1012" [TFF 6 0]
...
```

DEMO.DWG:
```
=======> Second Header 3 (start):   191443
before_sentinel: 0xd9240000690e00006064250026a193044004000004320000000000000064000000000000000002000000000000ffffffff00000000 [53]
size: 100 [RL 0]
address: 191443 [BL 0]
version: "AC1012" [TFF 6 0]
```

LINE1.DWG:
```
=======> Second Header 3 (start):    26013
before_sentinel: 0x000000006800000070852500201bfa02924d000004320000000000000064000000000000000002000000000000ffffffff00000000 [53]
size: 114 [RL 0]
address: 26013 [BL 0]
version: "AC1012" [TFF 6 0]
```

btw: https://github.com/LibreDWG/libredwg/blob/master/src/encode.c#L94
```
/* See ODA p263, p 25 UNKNOWN SECTION */
const unsigned char unknown_section[53]
    = { 0x00, 0x00, 0x00, 0x00, /* RL 0 */
        0x00, 0x00, 0x00, 0x00, /* RL num_objmap_entries + 3 */
        0xA7, 0x62, 0x25, 0x00, 0xF6, 0xAF, 0x25, 0x02, 0x3B, 0x04, 0x00,
        0x00, /* obj start_address as RL (off 16) */
        0x04, 0x32, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x64, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0xFF, 0xFF, 0xFF, 0xFF, 0x00, 0x00, 0x00, 0x00 };
```
not used, but this is the structure before 2ndheader

rurban, 2023-11-11:
I would directly name it UNKOWN section, because it is one

michal-josef-spacek, 2023-11-11:
no problem

michal-josef-spacek, 2023-11-13:
@rurban Updated https://github.com/LibreDWG/libredwg/pull/865:
BLANK.DWG:
```
num_sections: 3 [RL]
num_sections => 3
section[0].number:         0 [RC] AcDb:Header
section[0].address:       70 [RL]
section[0].size:         454 [RL]
section[1].number:         1 [RC] AcDb:Classes
section[1].address:      524 [RL]
section[1].size:          38 [RL]
section[2].number:         2 [RC] AcDb:Handles
section[2].address:     3977 [RL]
section[2].size:          79 [RL]
crc: A24C [RSx] from 0-52
         HEADER (end):      70
...
=======> Last Object      :     2255
last object size: 63 [MS] (@3910)
         Last Object (end):     3975

=======> Second Header (start):     4056
size: 90 [RL 0]
address: 4056 [BL 0]
version: "AC1012" [TFF 6 0]
...
```
DEMO.DWG:
```
num_sections: 4 [RL]
num_sections => 4
section[0].number:         0 [RC] AcDb:Header
section[0].address:       79 [RL]
section[0].size:         459 [RL]
section[1].number:         1 [RC] AcDb:Classes
section[1].address:      538 [RL]
section[1].size:          38 [RL]
section[2].number:         2 [RC] AcDb:Handles
section[2].address:   183733 [RL]
section[2].size:        7657 [RL]
section[3].number:         3 [RC] AcDb:ObjFreeSpace
section[3].address:   191390 [RL]
section[3].size:          53 [RL]
crc: B70C [RSx] from 0-61
         HEADER (end):      79
...
=======> Last Object      :     1431
last object size: 148 [MS] (@183581)
         Last Object (end):   183731

=======> Obj Free Space 3 (start): 191390
         Obj Free Space 3   (end): 191443
zero: 0x24D9 [RLL 0]
numhandles: 0xE69 [RLL 0]
TDUPDATE: [2450528, 76783910] 1997-03-20 18:25:41 [TIMERLL 0]
objects_address: 0x440 [RLx 0]
numnums: 0x4 [RC 0]
max32: 0x32 [RLL 0]
max64: 0x64 [RLL 0]
maxtbl: 0x200 [RLL 0]
maxrl: 0xFFFFFFFF [RLL 0]

=======> Second Header (start):   191443
size: 100 [RL 0]
address: 191443 [BL 0]
version: "AC1012" [TFF 6 0]
...
```
LINE1.DWG:
```
num_sections: 5 [RL]
num_sections => 5
section[0].number:         0 [RC] AcDb:Header
section[0].address:    18387 [RL]
section[0].size:         484 [RL]
section[1].number:         1 [RC] AcDb:Classes
section[1].address:    18871 [RL]
section[1].size:         475 [RL]
section[2].number:         2 [RC] AcDb:Handles
section[2].address:    25679 [RL]
section[2].size:         281 [RL]
section[3].number:         3 [RC] AcDb:ObjFreeSpace
section[3].address:    25960 [RL]
section[3].size:          53 [RL]
section[4].number:         4 [RC] AcDb:Template
section[4].address:    26159 [RL]
section[4].size:           4 [RL]
crc: 220A [RSx] from 0-70
         HEADER (end):      88
...
=======> Last Object      :    19858
last object size: 54 [MS] (@25621)
         Last Object (end):    25677

=======> Obj Free Space 3 (start): 25960
         Obj Free Space 3  (end): 26013
zero: 0x0 [RLL 0]
numhandles: 0x68 [RLL 0]
TDUPDATE: [2458992, 49945376] 2020-05-22 11:59:12 [TIMERLL 0]
objects_address: 0x4d92 [RLx 0]
numnums: 0x4 [RC 0]
max32: 0x32 [RLL 0]
max64: 0x64 [RLL 0]
maxtbl: 0x200 [RLL 0]
maxrl: 0xFFFFFFFF [RLL 0]

=======> Second Header (start):    26013
size: 114 [RL 0]
address: 26013 [BL 0]
version: "AC1012" [TFF 6 0]
...
```

michal-josef-spacek, 2023-11-13:
"Obj Free Space" section TODOs:
* zero variable isn't zero
* encoding

rurban, 2023-11-14:
ObjFreeSpace it is! 
very good catch, better than ODA

rurban, 2023-11-14:
My branch work/r13-objfreespace is leading to much more ODA Invalid input errors, than master.

michal-josef-spacek, 2023-11-14:
Seems ok, I need to test.

michal-josef-spacek, 2023-11-14:
@rurban 
decode.c: In function ‘decode_R13_R2000’:
decode.c:950:40: error: ‘pvz’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  950 |               || dat->byte == pvz + 53 + 16))

And we need to move constant 53 to define as previous version.

rurban, 2023-11-14:
Shouldn't we publish a revised ODA spec, with all our found doc bugs and missing pieces?

michal-josef-spacek, 2023-11-14:
@rurban Is ODA somehow working at this time? 
Do you have a link to ODA spec?

michal-josef-spacek, 2023-11-14:
@rurban I don't think that your logic in https://github.com/LibreDWG/libredwg/blob/work/r13-objfreespace/src/decode.c#L926 is right.

You are writing:
* if 2NDSENTINEL, then possible ObjFreeSpace section. 
This is not true.

There are possibilities in header.sections:
a) No ObjFreeSpace section
b) ObjFreeSpace section with address 0 and size 0
c) ObjFreeSpace section with real address and size.

in the case of a) and b) we don't have ObjFreeSpace section.
in the case of c) we have ObjFreeSpace section.

In the case of 2NDHEADER, this is independent of this header record.
This is related to 2NDHEADER_SENTINEL only.

michal-josef-spacek, 2023-11-14:
> @rurban decode.c: In function ‘decode_R13_R2000’: decode.c:950:40: error: ‘pvz’ may be used uninitialized in this function [-Werror=maybe-uninitialized] 950 | || dat->byte == pvz + 53 + 16))

I don't see what this mean :-D

michal-josef-spacek, 2023-11-14:
a) BLANK.DWG
b) [Dodge-Challenger-1973.dwg.gz](https://github.com/LibreDWG/libredwg/files/13352055/Dodge-Challenger-1973.dwg.gz)
c) DEMO.DWG and LINE1.DWG

rurban, 2023-11-24:
Fixed

michal-josef-spacek, 2023-11-24:
@rurban Thanks for the fixes and merge.

error with pvz is gone. Super.

I could test more.

michal-josef-spacek, 2023-11-24:
Issue 7) gone
Issue 9) is still present.

The issue:
I have line.dwgadd ([line.dwgadd.gz](https://github.com/LibreDWG/libredwg/files/13457908/line.dwgadd.gz))
command: ```dwgadd -v9 --as r14 -o line.dwg line.dwgadd```
Output:
```
=======> Second Header (start): 2408
size: 0 [RL 0]
address: 0 [BL 0]
version: "AC1014" [TFF 6 0]
414331303134
zero_5: { 0x0 0x0 0x0 0x0 0x0 } [*RC]
is_maint: 0x0 [RC 0]
zero_one_or_three: 0x0 [RC 0]
dwg_version: 21 [BS 0]
=> header dwg_version: 0x15
=> header maint_version: 0x0
codepage: 30 [RS 0]
num_sections: 5 [BS 0]
sections[0].nr: 0 [RCd 0]
 Header
sections[0].address: 125 [BL 0]
sections[0].size: 465 [BL 0]
sections[1].nr: 1 [RCd 0]
 Classes
sections[1].address: 590 [BL 0]
sections[1].size: 38 [BL 0]
sections[2].nr: 2 [RCd 0]
 Handles
sections[2].address: 2185 [BL 0]
sections[2].size: 154 [BL 0]
sections[3].nr: 3 [RCd 0]
 ObjFreeSpace
sections[3].address: 2339 [BL 0]
sections[3].size: 53 [BL 0]
sections[4].nr: 4 [RCd 0]
 Template
sections[4].address: 0 [BL 0]
sections[4].size: 0 [BL 0]
num_handles: 14 [BS 0]
handles[0].num_hdl: 8 [RCd 0]
handles[0].nr: 0 [RCd 0]
hdl[0]: 0x25 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[0] HANDSEED: 0.8.250000200
handles[1].num_hdl: 8 [RCd 0]
handles[1].nr: 1 [RCd 0]
hdl[0]: 0x1 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[1] BLOCK_CONTROL_OBJECT: 0.8.10000200
handles[2].num_hdl: 8 [RCd 0]
handles[2].nr: 2 [RCd 0]
hdl[0]: 0x2 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[2] LAYER_CONTROL_OBJECT: 0.8.20000200
handles[3].num_hdl: 8 [RCd 0]
handles[3].nr: 3 [RCd 0]
hdl[0]: 0x3 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[3] STYLE_CONTROL_OBJECT: 0.8.30000200
handles[4].num_hdl: 8 [RCd 0]
handles[4].nr: 4 [RCd 0]
hdl[0]: 0x5 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[4] LTYPE_CONTROL_OBJECT: 0.8.50000200
handles[5].num_hdl: 8 [RCd 0]
handles[5].nr: 5 [RCd 0]
hdl[0]: 0x6 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[5] VIEW_CONTROL_OBJECT: 0.8.60000200
handles[6].num_hdl: 8 [RCd 0]
handles[6].nr: 6 [RCd 0]
hdl[0]: 0x7 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[6] UCS_CONTROL_OBJECT: 0.8.70000200
handles[7].num_hdl: 8 [RCd 0]
handles[7].nr: 7 [RCd 0]
hdl[0]: 0x8 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[7] VPORT_CONTROL_OBJECT: 0.8.80000200
handles[8].num_hdl: 8 [RCd 0]
handles[8].nr: 8 [RCd 0]
hdl[0]: 0x9 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[8] APPID_CONTROL_OBJECT: 0.8.90000200
handles[9].num_hdl: 8 [RCd 0]
handles[9].nr: 9 [RCd 0]
hdl[0]: 0xa [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[9] DIMSTYLE_CONTROL_OBJECT: 0.8.A0000200
handles[10].num_hdl: 8 [RCd 0]
handles[10].nr: 10 [RCd 0]
hdl[0]: 0xb [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[10] VX_CONTROL_OBJECT: 0.8.B0000200
handles[11].num_hdl: 8 [RCd 0]
handles[11].nr: 11 [RCd 0]
hdl[0]: 0xc [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[11] DICTIONARY_NAMED_OBJECT: 0.8.C0000200
handles[12].num_hdl: 8 [RCd 0]
handles[12].nr: 12 [RCd 0]
hdl[0]: 0x17 [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[12] DICTIONARY_ACAD_MLINESTYLE: 0.8.170000200
handles[13].num_hdl: 8 [RCd 0]
handles[13].nr: 13 [RCd 0]
hdl[0]: 0xd [RC 0]
hdl[1]: 0x0 [RC 0]
hdl[2]: 0x0 [RC 0]
hdl[3]: 0x0 [RC 0]
hdl[4]: 0x0 [RC 0]
hdl[5]: 0x2 [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[13] DICTIONARY_ACAD_GROUP: 0.8.D0000200
crc: 5E8C [RSx]
Warning: check_CRC mismatch 16-2599 = 2583: 7E57 <=> 5E8C
```

There are bad ```size```, ```address```, bad template address and size, bad crc.
And handle HANDSEED is probably bad too.

michal-josef-spacek, 2023-11-24:
Another issue with the conversion from JSON to DWG in AC1012. Don't encode the second header and ObjFreeSpace.

michal-josef-spacek, 2023-11-30:
In actual master

Conversion of JSON artefact
[first.json.gz](https://github.com/LibreDWG/libredwg/files/13513972/first.json.gz)
to DWG doesn't encode 2nd header.

michal-josef-spacek, 2023-12-12:
@rurban 
There is issue with decoding of second header in actual master.
On BLANK.DWG:
```
=======> Second Header (start): 4072
size: 90 [RL 0]
address: 4056 [BL 0]
ERROR: Invalid version size 6. Need min. 48 bits, have -32622 for .
         Second Header (end)  : 4178
```

Problem is in ```_VECTOR_CHKCOUNT_STATIC``` in ```FIELD_TFF``` (https://github.com/LibreDWG/libredwg/blob/master/src/dec_macros.h#L504)

michal-josef-spacek, 2023-12-13:
PR for simplifying: https://github.com/LibreDWG/libredwg/pull/894

michal-josef-spacek, 2023-12-24:
@rurban 
The problem with:

```
=======> Second Header (start): 4072
size: 90 [RL 0]
address: 4056 [BL 0]
ERROR: Invalid version size 6. Need min. 48 bits, have -32622 for .
         Second Header (end)  : 4178
```

is related to commit https://github.com/LibreDWG/libredwg/commit/10c79fd51df35afdc3e0272e58075e8d5d2e9a3c

Where is new:
```
// for error logging only:
Dwg_Object *obj = &(Dwg_Object){ .name = (char*)"2NDHEADER" };
```

Which define obj variable and then fail macro ```AVAIL_BITS```, because obj->size is 0.

rurban, 2023-12-24:
Aah! This explains it

michal-josef-spacek, 2023-12-25:
@rurban Thanks for code change.

There is issue still.

There is another construct ```bit_position```:
```
// check for overflow into next object (invalid num_elems)
#define AVAIL_BITS(dat)                                                       \
  (obj ? (int64_t)((obj->size * 8) - bit_position (dat) + 20)                 \
       : (int64_t)0x0000ff00LL)
```
which is failing. 

This AVAIL_BITS expects dat structure which is this block only.
This is not true in this situation.

rurban, 2023-12-27:
Should be fixed now, but haven't cross-checked yet

michal-josef-spacek, 2023-12-31:
@rurban There is issue still.

rurban, 2023-12-31:
Ok, I'll check

michal-josef-spacek, 2024-01-10:
In branch section_order decoding is ok.
The ```dwgrewrite``` doesn't encode the second header.

rurban, 2024-01-11:
No, in branch section_order json roundtrips still failing, i.e. the encoder for the 2ndheader, and some section addresses are missing.

rurban, 2024-01-29:
Should be fixed with ebfebfdb2b44cb83485f64b8699c87c5656618f8

rurban, 2024-01-30:
nope, fixed with 6967472446ed48a9b07ec183c9eac9b2e4c1505e

michal-josef-spacek, 2024-01-30:
1)
dwgrewrite on this test file doesn't work:
[BLANK.DWG.gz](https://github.com/LibreDWG/libredwg/files/14101924/BLANK.DWG.gz)
[BLANK.DXF.gz](https://github.com/LibreDWG/libredwg/files/14101926/BLANK.DXF.gz)

2)
I think that the second header size isn't right, but I need to clarify.
Seems that size is count of data from sentinel to sentinel. After dwgrewrite this is not right.

3)
The writing of the second header via dwgadd is not working.
Because bad writing of handle like:
```
...
num_handles: 14 [BS 0]
handles[0].num_hdl: 8 [RCd 0]
handles[0].nr: 0 [RCd 0]
hdl[0]: 0x25 [RC 0]
hdl[1]: 0xa0 [RC 0]
hdl[2]: 0xb9 [RC 0]
hdl[3]: 0xa7 [RC 0]
hdl[4]: 0xfc [RC 0]
hdl[5]: 0x7f [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[0] HANDSEED: 0.8.25A0B9A7FC7F00
handles[1].num_hdl: 8 [RCd 0]
handles[1].nr: 1 [RCd 0]
hdl[0]: 0x1 [RC 0]
hdl[1]: 0xa0 [RC 0]
hdl[2]: 0xb9 [RC 0]
hdl[3]: 0xa7 [RC 0]
hdl[4]: 0xfc [RC 0]
hdl[5]: 0x7f [RC 0]
hdl[6]: 0x0 [RC 0]
hdl[7]: 0x0 [RC 0]
[1] BLOCK_CONTROL_OBJECT: 0.8.1A0B9A7FC7F00
...
```

michal-josef-spacek, 2024-01-31:
@rurban Question about header.sections and header.num_sections:
If I look to code I understand these facts:
* header.sections is the value from the header and it's related to a number of section tables in the header and second header.
* header.num_sections is the value computed and it's related to a number of all sections (in the case of R13-R2000 to header.sections + thumbnail + second header)

It is true?

Actual situation:
num_sections is in some code the same as sections
in some code + thumbnail only.

I have some code to fix this situation, but I don't understand intention.

michal-josef-spacek, 2024-01-31:
PR fixes only 1).

rurban, 2024-01-31:
> It is true?

Yes it is.
I think I've also found one more bug there. Just a bit too busy with other stuff

rurban, 2024-02-02:
Oh, I see it now

rurban, 2024-02-04:
If there are remaining bugs, lets make a new ticket, please

closed 2024-02-04 by rurban