LibreDWG/libredwg #864: Issue with DWG->JSON->DWG conversion of DICTIONARYWDFLT object in r13

michal-josef-spacek, 2023-11-09:
Example:
[first.dwg.gz](https://github.com/LibreDWG/libredwg/files/13309860/first.dwg.gz)
[first.json.gz](https://github.com/LibreDWG/libredwg/files/13309861/first.json.gz)
[second.dwg.gz](https://github.com/LibreDWG/libredwg/files/13309862/second.dwg.gz)

First.dwg is origin:
```
Next object: 12 Handleoff: 0x1 [UMC] Offset: -2952 [MC] @20335
==========================================
Object number: 12/C, Size: 53 [MS], Type: 500 [BS], Address: 20337
Add object DICTIONARYWDFLT [12] Decode object DICTIONARYWDFLT
handle: 0.1.E [H 5]
EED[0] size: 20 [BS]
EED[0] handle: 5.1.12
EED[0] raw: 20
EED[0] code: 0 [RC], string: len=13 [RC] cp=28 [RS_BE] "TREAT_AS_HARD" [TF]
EED[1] code: 70 [RC], short: 0 [RS]

EED[2] size: 0 (end)
num_eed: 2
bitsize: 350 [RL]
 Hdlsize: 0x4A, hdl_dat: @43.6 - @53.0 (53)
num_reactors: 1 [BL 0]
ownerhandle: (12.1.2) abs:12 [H 330]
reactors[0]: (4.1.C) abs:12 [H* 330] => DICTIONARY
xdicobjhandle: (3.0.0) abs:0 [H 360]
--common_size: 246
numitems: 1 [BL 0]
TV-ZERO 1
texts[0]: "" [TV 3]
 handle stream: +58 @36.4 MISSING (@43.6  @48.6)
defaultid: (2.1.F) abs:15 [H 340]
 padding: +18 MISSING
 object_map{E} = 12
 padding: 3D/1 (2 bits)
crc: 2A2B [RSx]
 check_CRC 20335-20390 = 55: 2A2B == 2A2B
```

After conversion to JSON, first.json:
```
    {
      "object": "DICTIONARYWDFLT",
      "dxfname": "ACDBDICTIONARYWDFLT",
      "index": 12,
      "type": 500,
      "handle": [0, 1, 14],
      "size": 53,
      "bitsize": 350,
      "eed": [
        {
          "size": 20,
          "handle": [5, 1, 18],
          "code": 0,
          "value": "TREAT_AS_HARD"
        },
        {
          "code": 70,
          "value": 0
        }
      ],
      "ownerhandle": [12, 1, 2, 12],
      "reactors": [
        [4, 1, 12, 12]
      ],
      "numitems": 1,
      "items": {
    
      },
      "_subclass": "AcDbDictionaryWithDefault",
      "defaultid": [2, 1, 15, 15]
    },
```

After conversion back to DWG, in second.dwg:
```
Next object: 12 Handleoff: 0x1 [UMC] Offset: 41 [MC] @19725
==========================================
Object number: 12/C, Size: 41 [MS], Type: 500 [BS], Address: 19727
Add object DICTIONARYWDFLT [12] Decode object DICTIONARYWDFLT
handle: 0.1.E [H 5]
EED[0] size: 20 [BS]
EED[0] handle: 5.1.12
EED[0] raw: 20
EED[0] code: 0 [RC], string: len=13 [RC] cp=28 [RS_BE] "TREAT_AS_HARD" [TF]
EED[1] code: 70 [RC], short: 0 [RS]

EED[2] size: 0 (end)
num_eed: 2
bitsize: 266 [RL]
 Hdlsize: 0x3E, hdl_dat: @33.2 - @41.0 (41)
num_reactors: 1 [BL 0]
ownerhandle: (12.1.2) abs:12 [H 330]
reactors[0]: (4.1.C) abs:12 [H* 330] => DICTIONARY
xdicobjhandle: (3.0.0) abs:0 [H 360]
--common_size: 246
numitems: 0 [BL 0]
defaultid: (2.1.F) abs:15 [H 340]
 padding: +6
 object_map{E} = 12
 padding: FF/3F (6 bits)
crc: A156 [RSx]
 check_CRC 19725-19768 = 43: A156 == A156
```

michal-josef-spacek, 2023-11-09:
As I understand, the issue is the conversion of ```texts[0]: "" [TV 3]``` to blank line in JSON items.
Sure?

rurban, 2023-11-09:
Should be easy to fix

rurban, 2023-11-24:
A wrong version guard, was SINCE R_2000. Fixed

closed 2023-11-24 by rurban

michal-josef-spacek, 2023-11-24:
great, it's working